Ultimaker/Cura #12760: cura 5.1 or 5.0 and 4.xx color scheme line type always in compatibility mode

mttr402, 2022-07-14:
### Application Version

5.0 or 5.1 or 4.13 etc

### Platform

linux mint 20.3 UNA

### Printer

ender 6

### Reproduction steps

Release Linux Mint 20.3 Una 64-bit
Kernel Linux 5.4.0-91-generic x86_64
MATE 1.26.0
[cura.log](https://github.com/Ultimaker/Cura/files/9108092/cura.log)

start cura doesn't matter which version 
go to preview mode
color scheme line type 
Show up as compatibility mode and no starts  (white dots visible)

### Actual results

Always in compatibility mode no matter what

### Expected results

should be able to change it even though "Force layer view compatibility mode" is not check off

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

log file included
[cura.log](https://github.com/Ultimaker/Cura/files/9108143/cura.log)

Asterchades, 2022-07-14:
The log is reporting that only OpenGL 3.1 is available. 4.1 is required to use anything other than the compatibility view, itself requiring 2.0 to function at all.

Unlike the Windows logs, however, it is not reporting what actual graphics hardware and software you are using. Could you provide those details yourself?

mttr402, 2022-07-14:
This is my graphic hardware. as for the software it's linux mint 20.3 with Desktop being Mate 1.26
not sure if opengl 4.1 is even available for 20.3

System:    Kernel: 5.4.0-91-generic x86_64 bits: 64 compiler: gcc v: 9.3.0 Desktop: MATE 1.26.0 
           wm: marco dm: LightDM Distro: Linux Mint 20.3 Una base: Ubuntu 20.04 focal 

CPU:       Topology: Quad Core model: Intel Core i7-2600 bits: 64 type: MT MCP arch: Sandy Bridge 
           rev: 7 L2 cache: 8192 KiB 
           flags: avx lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 54273 
           Speed: 3595 MHz min/max: 1600/3800 MHz Core speeds (MHz): 1: 3582 2: 3582 3: 3774 
           4: 3609 5: 3536 6: 3533 7: 3685 8: 3565 

Graphics:  Device-1: Intel 2nd Generation Core Processor Family Integrated Graphics 
           vendor: Micro-Star MSI driver: i915 v: kernel bus ID: 00:02.0 chip ID: 8086:0102 
           Device-2: AMD Juniper PRO [Radeon HD 5750] vendor: Hightech Information System 
           driver: radeon v: kernel bus ID: 02:00.0 chip ID: 1002:68be 
           Display: x11 server: X.Org 1.20.13 driver: ati,modesetting,radeon unloaded: fbdev,vesa 
           compositor: marco resolution: 1680x1050~60Hz, 1920x1080~60Hz 
           OpenGL: renderer: AMD JUNIPER (DRM 2.50.0 / 5.4.0-91-generic LLVM 12.0.0) 
           v: 3.3 Mesa 21.2.6 compat-v: 3.1 direct render: Yes