Ultimaker/Cura #12761: "Multiply selected" does not work for a single copy

TheRealSaeba, 2022-07-14:
### Application Version

5.0.0-beta+1

### Platform

Windows 10

### Printer

Artillery Sidewinder X1

### Reproduction steps

Selecting an object and "multiply Selected" in the context menu

### Actual results

Selecting "number of copies" with value 1 does not result in the object duplicated.

### Expected results

Selecting "number of copies" with value 1 should result in one copy of the object created (resulting in two objects in total)  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Selecting "number of duplicates" with a value of 2 and above works as intended

nallath, 2022-07-14:
This has been fixed for 5.1

closed 2022-07-14 by nallath

fieldOfView, 2022-07-14:
This was only an issue in the 5.0 beta 1. It was already fixed in the final 5.0 release.