Ultimaker/Cura #12762: 100% fill cubic subdivision is not solid, but contains many small divots

gbarbehenn, 2022-07-14:
### Application Version

5.0.0

### Platform

windows 10

### Printer

Monoprice Maker Select V2/Wanhao Duplicator I3

### Reproduction steps

Happens on all prints, the attached picture is for prints that I had reported other issues on.  Set infill to 100%, cubic subdivision.  The print contains an array of small divots, these, in turn produce surface defects, as can be seen in the picture.
I have not seen this on other fills, but then I print either solid or cubic subdivision.
[image]
?

### Actual results

As can be seen in the pictures, many small triangular divots, that damage the top and bottom walls.

### Expected results

100% fill, and no divots on the top and bottom walls.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Side walls do not appear to be affected, only top, bottom and interior.

GregValiant, 2022-07-14:
Where are the top and bottom surfaces?  The infill looks uncovered.
When you set the infill to 100% Cura goes to "lines" since it doesn't matter (solid through-and-through doesn't care what infill type you use).
Without a project file ("File | Save Project") it's impossible to trouble shoot this.  Create the project file and then Zip it and post the zip file here.

Ghostkeeper, 2022-07-18:
Infill patterns that have overlapping lines (like Cubic) tend to perform badly at high infill densities. Cura generates enough line volume to fill the infill region, but when the lines overlap, the flow rate gets interrupted when the nozzle passes over a previously printed line.

Not all combinations of settings are expected to print well. I suggest to use the Lines pattern instead.

If you can share a project file, we can verify that in the top layers of the infill it does extrude enough to fill the object in theory. Of course, Cubic Subdivision by its nature is meant to leave the inside of the model open to save filament, which may result in a lower density too (but it should only happen in the middle of a large infill volume, not towards the edges of the infill).

closed 2022-07-29 by github-actions[bot]