Ultimaker/Cura #12765: Variable line width exceeds speed / flow limits

akshimassar, 2022-07-15:
### Application Version

5.1.0-beta

### Platform

Arch Linux

### Printer

ZAV printer

### Reproduction steps

Slice a model. Try various settings for `Flow Equalization Ratio`

### Actual results

At 100% equalization it hits 168.35 mm/s speed. This is above max speed for my printer.
At 0% equalization it hits 26 mm^3/s. This is above extruder capability of my printer.
[image]
[image]

### Expected results

I would expect slicer to not exceed desired limits. Exceeding desired flow could clearly ruin the print, we are talking about 25% more in that example. It will result in underextrusion in the best case or filament jamming.

As for exceeding speed it's not so bad but for instance it also happens at the first layer of the print where low speed is essential for proper bed adhesion. Nozzle speed is probably not so important for plastic to plastic adhesion but some materials still require low speed for proper adhesion. Also tall prints usually require not exceeding certain speeds to prevent de-laminating them from the base. So basically I would like my configured speed to be treated as maximum speed unless otherwise specified.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[test.3mf.zip](https://github.com/Ultimaker/Cura/files/9120396/test.3mf.zip)
That's not the best example obviously but it could illustrate the point.

Now I have to check flow and speed for every print and manually decrease speed or line size to fit my printer and material. By doing so I slow down the entire print.

akshimassar, 2022-07-15:
That's might be better example. Simple cube 1.3 mm * 20 mm * 20 mm in size. Desired print speed is 100 mm/s (including outer wall). Real printing speed is 153.85 mm/s. Surface quality will be worse and I would expect to see extrusion inconsistency due to acceleration / deceleration.

[image]

Ghostkeeper, 2022-07-15:
The speed and line width settings indicate the default case, but can be changed for various reasons. The speed will change depending on the Minimum Layer Time. It will change because of the initial layer speed. Or because of bridging. And in this case, because of compensating for flow changes. It is not meant to be a "maximum speed" or anything. This is a misunderstanding.

Is there any way in which we can make it more clear that the speed is not a maximum speed?

Ghostkeeper, 2022-07-15:
In order to change the line width, Cura can either adjust the speed, or adjust the flow rate, or a combination of both. With the flow compensation set to 100%, it will adjust the speed only.

If Cura is not allowed to change the speed, and also not allowed to change the flow rate, then logically it must also not be allowed to change the line width.

Cura 5 will never leave any gaps between parallel walls any more. But it can get close by setting the Wall Transitioning Threshold Angle to 1 degree. But there's a better way to get your desired behaviour: Don't allow wider lines, but do allow thinner lines. I'd suggest you try the following settings:
* Set Minimum Even Wall Line Width to 0mm.
* Set Minimum Odd Wall Line Width to 0mm.
* Set the Flow Equalization Ratio to 0%.

The first two will ensure that the lines will only be made smaller to fit, never larger. After all, if it were to use a larger line than the desired flow would allow, this would exceed your maximum flow rate. With setting the minimum line width to 0mm, it will always be allowed to add an extra wall instead. The third one will ensure that with these thinner lines, it will not create a faster movement speed, thus meeting your expectations.

Does this solve your problem?

akshimassar, 2022-07-15:
> The speed and line width settings indicate the default case, but can be changed for various reasons.
Yet the only case where it is actually _increased_ is variable line width. Why are you assuming it is safe to raise speed or flow in that case?

> If Cura is not allowed to change the speed, and also not allowed to change the flow rate, then logically it must also not be allowed to change the line width.
That was not my point, I just  don't understand the assumption that raising flow / speed is safe to do. I'm trying to illustrate that it's not safe to increase these values and it's probably decrease them.

I will try your suggestion a bit later

GregValiant, 2022-07-16:
Hello @akshimassar 
I opened your original "test.3mf file and sliced it a couple of times.  I have a macro for MSExcel for analyzing gcode files and took this opportunity to tweak the macro.  (Please note that It does not use Accel or Jerk in it's calculations.)
Here are my results.

With "Flow Equalization Ration" at 0%:
Maximum Feed = 100mm/sec
Maximum Flow = 30mm³/sec
Average print speed of the model (ignoring the Initial Layer speed of 30) = 100mm/sec

With "Flow Equalization Ration" at 50%:
Maximum Feed = 131mm/sec
Maximum Flow = 39.9mm³/sec
Average print speed of the model (ignoring the Initial Layer speed of 30) = 100.47mm/sec

With "Flow Equalization Ration" at 100%:
Maximum Feed = 166mm/sec
Maximum Flow = 49.8mm³/sec
Average print speed of the model (ignoring the Initial Layer speed of 30) = 103mm/sec

At an acceleration of 6000mm/sec² it takes .83mm to reach 100mm/sec.  The length of the extrusions that show those high speeds are in the range of 0.0009mm to 0.16mm.  The extremely short length of those extrusions means there is no way for the print head to approach the speeds that Cura assigned and consequently the calculated flow numbers can't be hit either.
In a line .0009 long the max speed reached at accel = 6000 is about 3.25mm/sec.  (That is still ignoring the Jerk factor).

This brings up another question.  With your Maximum Resolution at 0.5 I would have expected that those short line segments would have been combined with other segments.  I don't think a line segment of .0009mm was supposed to get through.  That may be the key to understanding the wild Feed Rate numbers, and equally wild Flow Rate numbers.

akshimassar, 2022-07-16:
> Does this solve your problem?
Nope, with the settings you mentioned I'm still getting up to 25.16 flow
[image]

> This brings up another question. With your Maximum Resolution at 0.5 I would have expected that those short line segments would have been combined with other segments. I don't think a line segment [...]
That's kinda weird...

akshimassar, 2022-07-17:
With odd and even line sizes set to 0 I even got 29 cubic mm per second flow instead of 20. That seems unreasonable.

[image]

[flow_29.3mf.zip](https://github.com/Ultimaker/Cura/files/9128666/flow_29.3mf.zip)

GregValiant, 2022-07-18:
When I generated a gcode file I don't see that.  The highest flow in my gcode was 20 and the average speed of extrusions (not counting initial layer) was 97.15.
When the Z gets up to the three rods sticking up at Layer 53 the feed rate falls off to 10mm/sec and the flow rate drops to 4mm³/sec as Minimum Layer Time comes into play.
I did see the high flowrates in the initial file you posted.  The total length of extrusion for those high flow rates was 3.4mm and they also coincided with the high feedrates.

akshimassar, 2022-07-18:
I used cura 5.1.0-beta for flow_29.3mf
I re-started cura, re-opened project file and got the same result with flow up to 29 on the second layer today.

Ghostkeeper, 2022-07-18:
Thanks for your new project file.

It seems that it's not allowed to add another line (reducing the line width) because it's also limited by the number of walls it's allowed to use. In your case, 3 per side. In your screenshot, it's showing 6 wall lines per piece. It's not allowed to add another wall there and so it cannot reduce the line width of the other walls. Raising the number of walls allows adding more walls to these pieces, which allows the other walls to become thinner. For instance, raising it to 5 limits the flow rate to 20.00. If this setting change is undesirable for other parts of the print, I suggest that you use a cutting mesh.

The premise of this issue is still not the expected behaviour. The line width, speed and layer thickness do not imply a "maximum" flow rate. And Cura is expected to exceed the implied flow rate because it is expected to exceed the nominal line width depending on the geometry.

I'd change this to a feature request then, but really this is then a duplicate of the request to implement a maximum flow rate. Because there is no maximum flow rate at the moment, indeed. So I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/5248.

closed 2022-07-18 by Ghostkeeper

akshimassar, 2022-07-18:
Ok then. I guess there is nothing I can do if you don't see increasing speed and flow as potential risk.

GregValiant, 2022-07-18:
What software are you using to get the Flow Numbers and the Feed Rate numbers?
Does that software take into account "Variable Line Width", "Acceleration", and "Jerk"?
When looking at a "discrete" section of Gcode those values must be taken into account.

Feed Rate and is expressed in the "F" parameter of a Gcode line.  The Accel and Jerk affect the Feed Rate at any given point along a line of an extrusion.  There is the accel portion, then the portion at "Feed Rate", and finally the decel portion.  The start speed is often the Jerk value (rather than 0) and the end speed is often the Jerk value.
If your software is simply expressing the "F" parameter of a line, then the "Speed across the Distance" is wrong as it is ignoring Accelerate-Move-Decelerate.  This will cause errors.  The speed of an extrusion may be called out as 100, but the average speed across the extrusion will never be 100.
If your software is using that "wrong speed" and/or using the default Line Width (rather than variable line width), then the Flow Rate is wrong as well.

Consider these two lines of Gcode taken from the "test.gcode" I generated from your project file:
G1 F9345.8 X38.709 Y47.404 E3261.17384
G1 F9983.4 X39.679 Y47.542 E3261.2228
Taking Accel and Jerk into account, and using the volume that was actually extruded vs. the volume that should have been extruded at a Line Width of 1.0mm, I get these numbers:
Extrusion Length = 0.97mm
(Layer Height from settings) = 0.20mm
Extruded Volume = .11775mm³
Expected Volume (at a Line Width of 1.0) = 0.19446
Line Width = (.11775/.97)/0.20 = 0.607mm
Expected Line Width if Expected Flow is used = (.19446/.97)/0.20 = 1.002
Average Speed = 43mm/sec
Flow Rate = 5.07mm³/sec
Flow Rate using 1.0, 0.20, at 100mm/sec = 20mm³/sec
Flow Rate using the "F" parameter and the Expected Volume = 32.35mm³/sec

This is the Inner Wall area that contains those two lines of gcode.
[image]

This image is a detail view inside the circle.  The dimensions indicate the length of the extrusion lines and the ".6105" dimension indicates the distance between two adjacent extrusions.  "0.6105" is the Line Width actually used in the print at that exact point.  You can see that it is very close to my calculated line width of 0.607 obtained by working backwards through the numbers.
[image]

When doing a discrete analysis of a particular point then it is necessary to use exact numbers.
When looking at the overall numbers of a file then it doesn't matter as everything tends to smooth out.
Though the entire Gcode file, and ignoring the slower first layer:
"F" parameter Average Extrusion Speed = 101.66mm/sec
Average Flow Rate = 20.76

I don't really see a bug here.  There are numbers in the file that seem out of whack, but when looked at closely they appear to have been calculated because Cura is making a concerted effort to keep the speed and flow nominal based on the other numbers which are moving targets.

akshimassar, 2022-07-18:
> What software are you using to get the Flow Numbers and the Feed Rate numbers?
I'm using Cura if that's a question for me.

I'm not getting you point, sorry. I see up to 1.5x increase in speed or in flow on prolonged lines. Long lines happen when you have more _square_ forms with parallel edges thus Cura have to fill in the gap between those lines. And I'm 100% confident 1.5x increase in flow will ruin some prints, not so confident about speed increase -- usually you can get away with it. In my opinion slicer should never exceed configured limits unless explicitly specified. De facto lowering speed or flow is considered to be always safe and known to mitigate many printing issues.

Ultimaker team however choose to increase those values in order to get you faster prints. A couple percent faster probably. Ruining experience for those who already print at the edge of printer or material capabilities.

Here is simple example, cube with 4.5 mm width and 1.0 mm line, 3 walls. Cura decides 0.5 mm is not good enough and fills it, resulting in 1.25 mm line width.
[image]