Ultimaker/Cura #12771: Model won't slice

brekkeso, 2022-07-15:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 5 Pro (Custom created-default had zero quality profiles)

### Reproduction steps

Put 3 stl files on bed (NM, PA, NC-2, from https://www.thingiverse.com/thing:1524543)
Scaled to 50%
Rotated 90degrees on X or Y (models were not positioned flat)
Adjusted Z so majority of bottom surface flat on bed (default put them either under bed or above as overhang).
Clicked Slice. 

### Actual results

Received error.
[image]

### Expected results

Should have sliced the file into gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[NC-2-PA-NM.zip](https://github.com/Ultimaker/Cura/files/9124103/NC-2-PA-NM.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9124122/cura.log)

Asterchades, 2022-07-16:
So this is a curious one. All three models could use some repairs but doing so does not fix the issue. Loading each STL individually and following the same steps also doesn't produce the issue, nor does manually arranging them all onto the build platform after loading them individually.

Working from the provided project file backwards, NC-2 (object "NC-2-PA-NM.3mf" in the project) is causing the problem. Moving it causes no change, however both scaling and rotating it do. Adjusting the scale ±1% lets it slice correctly, as does rotating it by any value that is not a multiple of 90°. Working back from the original STL shows the same issues if scaled and oriented appropriately when using the same profile.

It's that last part that seems to be the deciding factor - the print profile. When using my own profile it slices successfully. When using the default Draft profile it slices successfully. And even recreating the 3D Topo Map profile setting by setting still lets it slice successfully. It's only when using the profile as configured in the project that this specific size and orientation combination causes an issue.

Flagging this one so hopefully a Dev can take a better look at it than I can.

Ghostkeeper, 2022-07-18:
Yes. It was hitting an edge case in the thin pieces in that NC-2 object. Cura 5.0 had an edge case where thin pieces with almost-parallel opposite walls would sometimes give this crash. It was an edge case, so changing the geometry slightly (by scaling it, or by rotating it such that rounding errors fall differently, or by changing certain settings) can prevent the issue when the geometry it generates walls with is different.

We've made a fix in 5.1 and now it slices correctly.

[image]

Thanks for the report!

closed 2022-07-18 by Ghostkeeper