Ultimaker/Cura #12772: Horizontal expansion and initial layer horizontal expansion fields swapped

dsdanielko, 2022-07-15:
### Application Version

5.0.1 beta

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

Changed horizontal expansion value -> Slice -> Preview
Changed initial horizontal expansion value -> Slice -> Preview

### Actual results

Changed horizontal expansion value -> Slice -> Preview and saved gcode behaves as if initial horizontal expansion value was changed
Changed initial horizontal expansion value -> Slice -> Preview and saved gcode behaves as if horizontal expansion value was changed

### Expected results

Changed horizontal expansion value -> Slice -> Preview and saved gcode behaves as if horizontal expansion value was changed
Changed initial horizontal expansion value -> Slice -> Preview and saved gcode behaves as if initial horizontal expansion value was changed

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_ceiling_door_hook.zip](https://github.com/Ultimaker/Cura/files/9124171/CE3PRO_ceiling_door_hook.zip)
[image]
[image]
[image]
[image]

GregValiant, 2022-07-16:
Than you for the report.
This is a duplicate of about "a lot" of bug reports.  I'll mark it as a duplicate and close it.  If you think that is premature let me know here.

closed 2022-07-16 by GregValiant

fieldOfView, 2022-07-16:
The good news is that it has been fixed, and the fix will be in Cura 5.1:
https://github.com/Ultimaker/CuraEngine/pull/1695