Ultimaker/Cura #12773: Running cura from source PyQt6.QtNetwork module not found

yess1987, 2022-07-16:
### Application Version

5.1.0

### Platform

windows 10

### Printer

no printer

### Reproduction steps

Windows
Windows 10 or higher
Visual Studio with MSVC 2019 or higher
Python 3.6 or higher
venv
CMake 3.20 or higher
Ninja 1.10 or higher
Conan 1.48 or higher

`pip install conan --upgrade
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
conan remote remove cura
powershell

cd C:\dev\cura_inst
.\Scripts\activate.ps1
 python cura_app.py`

Have tried the following，but it doesn't work。
“As a possible intermittent solution you can try to copy all the .dll file into the Scripts folder or in to the root of you Lib\site-packages”

### Actual results

Traceback (most recent call last):
  File "C:\dev\cura_inst\cura_app.py", line 23, in <module>
    from PyQt6.QtNetwork import QSslConfiguration, QSslSocket
ImportError: DLL load failed while importing QtNetwork: 找不到指定的模块。

### Expected results

i can run cura from soure

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

`pip install conan --upgrade
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
conan remote remove cura
powershell

cd C:\dev\cura_inst
.\Scripts\activate.ps1
 python cura_app.py`

yess1987, 2022-07-16:
Does “venv” need to be installed，I use python 3.10.x

Ghostkeeper, 2022-07-18:
Yes. Please see the instructions here: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

One of the requirements to be installed first is `venv`. Without a virtual environment, I'm not surprised that it can't find this dependency.

Let me know if installing that and repeating the instructions fixes your issue!

nallath, 2022-07-19:
> Let me know if installing that and repeating the instructions fixes your issue!

It reproduces for me on a clean windows install.

jellespijker, 2022-07-22:
This bug should be addressed by our CURA-8831 ticket  on the backlog.

yess1987, 2022-08-07:
i can find the venv by python3.10. Did I need install it
how to install the "venv", please give me a link for installing.

yess1987, 2022-08-07:
"This bug should be addressed by our CURA-8831 ticket on the backlog."

What is the current progress？

jellespijker, 2022-08-08:
I believe @casperlamboo is picking up this ticket today

Piezoid, 2022-08-08:
This is the error I get when I forget to activate the virtualenv.
When it is correctly activated, `python -c "import sys; print(sys.path)"` should contain en entry with `Cura/venv/lib/python3.10/site-packages` along with various paths under the conan repository

jellespijker, 2022-08-08:
@Piezoid are you on windows? Because the PyQt6.QtNetwork error seems to be only happening on Windows.
When I check the dependencies (with the Dependency walker) of the PyQt6 dll's and pyd's they seem to be be linking to a system Python: python3.dll while the CPython compiled by Conan generates a python310.dll.

But this was just a quick initial investigation.

@casperlamboo possible fixes for these are:
- compile Qt6 from source (there is a Conan recipe) and then compile PyQt6 linking against Conan
- it might be possible to link the pyd and dll to the Conan python310.dll, but I don't know how to do that on windows.
- change the name of the python310.dll to python3.dll in the Conan file, we currently use this modified recipe https://github.com/conan-io/conan-center-index/compare/master...jellespijker:conan-center-index:add_3_10_4_version_cpython
You might have to look into how CPython is compiled on windows https://github.com/python/cpython/blob/main/PCbuild/readme.txt

I'm tempted to consider this a bug on the PyQt6 side, since they link to another name then CPython provides by default (which is python310.dll) you could check if they have an open issue or maybe an newer version might fix this.

ClnSchlssr, 2022-08-12:
I'm running into the same issue but I don't totally understand the workaround here. I can't find any reference to python310.dll in the conanfile. Are there any other temporary fixes to try or should I just sit tight and raise the issue on the PyQt side?

jellespijker, 2022-08-27:
@ClnSchlssr There is no workaround. But we're working on a proper fix, which is involving building Qt6 and PyQt6 from source. This exact issue is fixed with: https://github.com/Ultimaker/conan-ultimaker-index/commit/6cd39f84e426b501a7b95588b0b034d91a138924 but that is fix incomplete because it doesn't fonts aren't working on Windows ( possible fix might be this PR https://github.com/conan-io/conan-center-index/pull/12368 ) and the file dialog on Linux also have some issues. But I'm slowly chipping away at those issues.

yess1987, 2022-09-22:
Has this issue been resolved?
@jellespijker

jellespijker, 2022-09-22:
Almost, I'm hoping that I can get my fixes ready for code review today. If not then one of my colleagues needs to pick this up for next week (due to me having a holiday and I made a promise to the missus to not look at GitHub 😅)

Bt the root cause has been found and fixed. It now needs to be fortified to work on all three OSes.

yess1987, 2022-09-23:
Is the code updated? Is it possible for me to update the code directly and then compile. But I see the compilation documentation is not updated
@jellespijker

jellespijker, 2022-09-23:
@rburema and/or @casperlamboo please update this ticket when finished

450083aa, 2022-09-26:
When is it expected to be repaired, is there an exact time? thanks!

yess1987, 2022-10-02:
Does this ticket  finished?

450083aa, 2022-10-08:
> When is it expected to be repaired, is there an exact time? thanks!

@jellespijker Is the vacation over?

eneserd11e, 2022-10-10:
i got same error. Please inform us. thanks

jellespijker, 2022-10-10:
yes, my holiday is finished.  Unfortunately my team members didn't get to finish it during my holiday and this sprint we're focused on the next stable release,  so I don't have time to work on this during my work hours.

However, I did some work on this during the weekend, I will cont. with it after working hours. No promises but..... I might have a fix today or tomorrow, depending on the grace of wife 😅

jellespijker, 2022-10-10:
I just pushed some of my changes, can you guys try again?
Please clear your Conan cache first and make sure you use the latest Conan version (>=1.53.0)

**Run all commands from Powershell**

Upgrade Conan and clean cache
```powershell
pip install conan --upgrade
conan remove "*" -f
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
```

Start from clean Cura situation
```powershell
cd Cura
git switch main
git pull
rmdir venv
```

Run the install commands
```powershell
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
```

Start Cura
```powershell
.\venv\Scripts\Activate.ps1
python cura_app.py
```

For more information see: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source for 

Let me know if this fixes things

ZacharyHowell-Essentium, 2022-10-10:
@jellespijker 
That seems to allow me to run from source. Great job!

eneserd11e, 2022-10-10:
Thanks for your assistance and interest thanks.it worked for me too. You are the best @jellespijker 😅

jellespijker, 2022-10-10:
Great to hear that!

I hope to see some nice Pull Requests coming in soon 😉

I will close this ticket as solved.

@Mariska fyi

yess1987, 2022-10-14:
Can be based on the 4.10 cura？, or what methods are available？
git clone https://github.com/Ultimaker/Cura.git

jellespijker, 2022-10-14:
> Can be based on the 4.10 cura？, or what methods are available？
> git clone https://github.com/Ultimaker/Cura.git 

for building Cura 4.x please take a look at the archived cura-build-environment and cura-build repositories.  please be aware that you are in for a hell of a ride and we can't offer any support. 

we moved away from that approach for a reason.

good luck

jellespijker, 2022-10-14:
> sipbuildtool/0.2.2@ultimaker/stable: ERROR: Traceback (most recent call last):
>   File "C:\Users\shuan\AppData\Local\Programs\Python\Python310\lib\site-packages\conans\client\generators\__init__.py", line 177, in write_generators
>     generator = generator_class(conanfile)
>   File "C:\Users\shuan\AppData\Local\Programs\Python\Python310\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 94, in __init__
>     self.configuration = conanfile.settings.build_type
>   File "C:\Users\shuan\AppData\Local\Programs\Python\Python310\lib\site-packages\conans\model\settings.py", line 279, in __getattr__
>     self._check_field(field)
>   File "C:\Users\shuan\AppData\Local\Programs\Python\Python310\lib\site-packages\conans\model\settings.py", line 275, in _check_field
>     raise undefined_field(self._name, field, self.fields, self._parent_value)
> conans.errors.ConanException: 'settings.build_type' doesn't exist
> 'settings' possible configurations are none
> 
> ERROR: Error in generator 'MSBuildDeps': 'settings.build_type' doesn't exist
> 'settings' possible configurations are none

it looks like you didn't install the latest conan version and our conan-config .

0lease follow the steps on our wiki: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

please be aware that this is only for our 5.2 and main branch. Earlier versions are not supported with this.

yess1987, 2022-10-14:
Does vs2022 must be installed? I use vs2019 now

yess1987, 2022-10-14:
ERROR: pyarcus/5.2.0-beta.2+1@ultimaker/stable: Error in generate() method, line 95
        sip.build()
while calling 'build', line 34
        self._conanfile.run(cmd)
        ConanException: Error 1 while executing "sip-build"

jellespijker, 2022-10-14:
> Does vs2022 must be installed? I use vs2019 now

the requirements in the wiki state 2022

jellespijker, 2022-10-14:
> ERROR: pyarcus/5.2.0-beta.2+1@ultimaker/stable: Error in generate() method, line 95 sip.build() while calling 'build', line 34 self._conanfile.run(cmd) ConanException: Error 1 while executing "sip-build"

please check the wiki for the requirements 

I will close those issue again, since the problems you encounter are unrelated to this issue

closed 2022-10-14 by jellespijker

yess1987, 2022-10-15:
i got it,thanks

alexandr-vladimirov, 2022-10-22:
facing the same error, conan version is 1.50.2

conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
curaengine:compiler.cppstd=20
umspatial*:compiler.cppstd=20
[options]
cura:devtools=True
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True

sipbuildtool/0.2.2@ultimaker/stable: Not found in local cache, looking in remotes...
sipbuildtool/0.2.2@ultimaker/stable: Trying with 'cura-ce'...
Downloading conanmanifest.txt completed [0.12k]
Downloading conanfile.py completed [1.57k]
sipbuildtool/0.2.2@ultimaker/stable: Downloaded recipe revision a636cf06c54e438248f657020656457d
pyarcus/5.3.0-alpha+testing_1@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.3.0-alpha while recipe is build for version: 5.3.0-alpha+testing_1
arcus/5.3.0-alpha+testing_5@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.3.0-alpha while recipe is build for version: 5.3.0-alpha+testing_5
WARN: arcus/5.3.0-alpha+testing_5@ultimaker/testing: requirement standardprojectsettings/[>=0.1.0]@ultimaker/stable overridden by pyarcus/5.3.0-alpha+testing_1@ultimaker/testing to standardprojectsettings/0.1.0@ultimaker/stable
WARN: cpython/3.10.4: requirement zlib/1.2.11 overridden by pyarcus/5.3.0-alpha+testing_1@ultimaker/testing to zlib/1.2.12
curaengine/5.3.0-alpha+testing_33@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.3.0-alpha while recipe is build for version: 5.3.0-alpha+testing_33
curaengine/5.3.0-alpha+testing_33@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.3.0-alpha while recipe is build for version: 5.3.0-alpha+testing_33
WARN: spdlog/1.10.0: requirement fmt/8.1.1 overridden by curaengine/5.3.0-alpha+testing_33@ultimaker/testing to fmt/9.0.0
pysavitar/5.3.0-alpha+testing_1@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.3.0-alpha while recipe is build for version: 5.3.0-alpha+testing_1
savitar/5.3.0-alpha+testing_0@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.3.0-alpha while recipe is build for version: 5.3.0-alpha+testing_0
WARN: savitar/5.3.0-alpha+testing_0@ultimaker/testing: requirement standardprojectsettings/[>=0.1.0]@ultimaker/stable overridden by pysavitar/5.3.0-alpha+testing_1@ultimaker/testing to standardprojectsettings/0.1.0@ultimaker/stable
pynest2d/5.3.0-alpha+testing_0@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.3.0-alpha while recipe is build for version: 5.3.0-alpha+testing_0
nest2d/5.2.0-alpha+testing_33@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.2.0-alpha while recipe is build for version: 5.2.0-alpha+testing_33
nest2d/5.2.0-alpha+testing_33@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.2.0-alpha while recipe is build for version: 5.2.0-alpha+testing_33
nest2d/5.2.0-alpha+testing_33@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.2.0-alpha while recipe is build for version: 5.2.0-alpha+testing_33
WARN: nest2d/5.2.0-alpha+testing_33@ultimaker/testing: requirement standardprojectsettings/[>=0.1.0]@ultimaker/stable overridden by pynest2d/5.3.0-alpha+testing_0@ultimaker/testing to standardprojectsettings/0.1.0@ultimaker/stable
uranium/5.3.0-alpha+testing_12@ultimaker/testing: WARN: Using dependencies specified in conandata.yml for version: 5.3.0-alpha while recipe is build for version: 5.3.0-alpha+testing_12
bzip2/1.0.8: WARN: Can't update, no package in remote
clipper/6.4.2: WARN: Can't update, no package in remote
cura_binary_data/5.3.0-alpha+testing_0@ultimaker/testing: WARN: Can't update, no package in remote
expat/2.4.1: WARN: Can't update, no package in remote
fdm_materials/5.3.0-alpha+testing_6@ultimaker/testing: WARN: Can't update, no package in remote
[... 138 lines cut ...]
nasm/2.15.05: Already installed!
nasm/2.15.05: Appending PATH environment variable: C:\.conan\42921a\1\bin
nlopt/2.7.0: Already installed!
pugixml/1.12.1: Already installed!
range-v3/0.12.0: Already installed!
rapidjson/1.1.0: Already installed!
Downloading conan_sources.tgz completed [1.14k]
Decompressing conan_sources.tgz completed [0.00k]
sipbuildtool/0.2.2@ultimaker/stable: Configuring sources in C:\.conan\696debb\1
sipbuildtool/0.2.2@ultimaker/stable: Copying sources to build folder
sipbuildtool/0.2.2@ultimaker/stable: Building your package in C:\.conan\d3bf1cb\1
sipbuildtool/0.2.2@ultimaker/stable: ERROR: Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\site-packages\conans\client\generators\__init__.py", line 173, in write_generators
    generator = generator_class(conanfile)
[... cut ...]
  File "C:\Program Files\Python310\lib\site-packages\conans\model\settings.py", line 266, in __getattr__
    self._check_field(field)
  File "C:\Program Files\Python310\lib\site-packages\conans\model\settings.py", line 262, in _check_field
    raise undefined_field(self._name, field, self.fields, self._parent_value)
conans.errors.ConanException: 'settings.build_type' doesn't exist
'settings' possible configurations are none

ERROR: Error in generator 'MSBuildDeps': 'settings.build_type' doesn't exist
'settings' possible configurations are none

jellespijker, 2022-10-23:
You're running into a Conan bug https://github.com/conan-io/conan/issues/11822 , which is fixed in the newer version of Conan.

running
```
pip install conan --upgrade
```
should do the trick

alexandr-vladimirov, 2022-10-27:
Omg, finally I did it, I was able to build and run Cura after several months after first try 🥳🥳🥳 Current instructions to build and run from source look much easier than before, thanks a lot for your efforts on this🙏

@jellespijker, **"pip install conan --upgrade"** magially helped because I tried before with the latest conan and it didn't work. On another laptop I could reproduce this magic: I installed the latest conan - faced the same error, downgraded it to 1.50.2 - still the same, upgraded to lates - and it worked.

After I started getting another error (below) which was able to solve by upgraging sip from 6.6.0 to latest 6.7.2 version. Current requirements for sip: **sip (Python) 6.6.x**, maybe it needs to be updated :)

pynest2d/5.3.0-alpha+testing_0@ultimaker/testing: Calling:
 > "sip-build"

----Running------
> "sip-build"
-----------------
These bindings will be built: pynest2d.
Generating the pynest2d bindings...
sip-build: C:\.conan\1cf6a3\1\python\BottomLeftConfig.sip: line 6 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\BottomLeftConfig.sip: line 24 column 9: '#' is unexpected
C:\.conan\1cf6a3\1\python\BottomLeftConfig.sip: line 31 column 1: '}' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 6 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 11 column 9: '#' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 19 column 11: exactly one character expected between single quotes
C:\.conan\1cf6a3\1\python\Box.sip: line 23 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 26 column 13: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 26 column 14: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 26 column 18: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 26 column 19: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 27 column 9: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 27 column 10: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 27 column 14: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 27 column 15: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 47 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 101 column 63: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 101 column 64: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 101 column 76: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 101 column 77: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 102 column 9: 'and' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 102 column 17: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 102 column 18: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 102 column 30: '`' is unexpected
C:\.conan\1cf6a3\1\python\Box.sip: line 102 column 31: '`' is unexpected
C:\.conan\1cf6a3\1\python\Circle.sip: line 4 column 7: classes, structs, unions and namespaces must be in the public or protected sections
C:\.conan\1cf6a3\1\python\Circle.sip: line 6 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\Circle.sip: line 11 column 9: '#' is unexpected
C:\.conan\1cf6a3\1\python\Circle.sip: line 21 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\Circle.sip: line 56 column 59: '*' is unexpected
C:\.conan\1cf6a3\1\python\DJDHeuristicConfig.sip: line 4 column 8: classes, structs, unions and namespaces must be in the public or protected sections
C:\.conan\1cf6a3\1\python\DJDHeuristicConfig.sip: line 6 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\DJDHeuristicConfig.sip: line 32 column 56: exactly one character expected between single quotes
C:\.conan\1cf6a3\1\python\DJDHeuristicConfig.sip: line 35 column 9: '#' is unexpected
C:\.conan\1cf6a3\1\python\DJDHeuristicConfig.sip: line 46 column 1: '}' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 6 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 26 column 9: '#' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 39 column 12: '!' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 43 column 9: class variables must be in the public section
C:\.conan\1cf6a3\1\python\Item.sip: line 44 column 27: '=' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 47 column 35: ';' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 50 column 26: '=' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 51 column 26: '=' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 59 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 63 column 59: exactly one character expected between single quotes
C:\.conan\1cf6a3\1\python\Item.sip: line 66 column 58: exactly one character expected between single quotes
C:\.conan\1cf6a3\1\python\Item.sip: line 107 column 72: 'the' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 121 column 72: 'the' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 151 column 5: '%' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 196 column 25: '.' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 197 column 70: exactly one character expected between single quotes
C:\.conan\1cf6a3\1\python\Item.sip: line 224 column 47: exactly one character expected between single quotes
C:\.conan\1cf6a3\1\python\Item.sip: line 257 column 65: exactly one character expected between single quotes
C:\.conan\1cf6a3\1\python\Item.sip: line 267 column 50: 'coordinate' is unexpected
C:\.conan\1cf6a3\1\python\Item.sip: line 277 column 48: 'coordinate' is unexpected
[... cut ...]
pynest2d/5.3.0-alpha+testing_0@ultimaker/testing: ERROR: Package '1800425ebafb2e3ff1a5e7e0c50b401d2cdeb407' build failed
pynest2d/5.3.0-alpha+testing_0@ultimaker/testing: WARN: Build folder C:\.conan\1cf6a3\1\build\Release
ERROR: pynest2d/5.3.0-alpha+testing_0@ultimaker/testing: Error in build() method, line 99
        cmake.configure()
        ConanException: Error 1 while executing cmake -G "Ninja" -DCMAKE_TOOLCHAIN_FILE="C:/.conan/1cf6a3/1/build/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="C:/.conan/095230/1" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Release" "C:\.conan\1cf6a3\1\."

jellespijker, 2022-11-03:
great to hear that you managed to run from source, we look forward to your PRs ;-) 

> After I started getting another error (below) which was able to solve by upgraging sip from 6.6.0 to latest 6.7.2 version. Current requirements for sip: sip (Python) 6.6.x, maybe it needs to be updated :)

This seems to be a bug in SIP which behaves differently in Python 3.10 and 3.11
Excellent that you figured out a fix for yourself. I'm not sure how to best describe the different version requirements in a uniform way for the thousands of different combinations of OSes, architectures, Python versions, compiler versions etc.

@MariMakes also had similar remarks on the instructions when she setup her environment maybe she has some suggestions for this