Ultimaker/Cura #12776: AppImage: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'

GAZ082, 2022-07-16:
### Application Version

5.1

### Platform

Linux x64/Archlinux

### Printer

Ender 3

### Reproduction steps

./Ultimaker-Cura_5.1.0-beta-linux.AppImag

### Actual results

PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

(Ultimaker-Cura:52974): GLib-GIO-ERROR **: 15:23:57.641: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'
/tmp/.mount_Ultimac2wMhn/AppRun: line 16: 52974 Trace/breakpoint trap   (core dumped) $scriptdir/Ultimaker-Cura "$@"

### Expected results

Open the program.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Trying to launch the latest linux appimage and got the error.

seven-sam, 2022-07-18:
the only solution found is to do this on fedora 36 :

sudo nano /etc/gdm/custom.conf

uncomment the line:
WaylandEnable=false

and add the line under [daemon]
DefaultSession=gnome-xorg.desktop

close your current session on the login page at the bottom right there is the setting icon choose Gnome-Xorg

launch Beta 5.1 like 5.0 with the preload command and it works

screenshot : https://i.postimg.cc/JM0qmbwk/Capture-d-cran-du-2022-07-18-23-35-22.png

GAZ082, 2022-07-23:
> Could you try again with this artefact? https://github.com/Ultimaker/Cura/suites/7402109420/artifacts/301727738
> 
> We don't officially support Wayland yet, as far as I know. We're having difficulty with that because the Qt Wayland plug-in is not ported with PyQt6 (yet).

Hi. I think that link got stale:
[image]

GAZ082, 2022-07-23:
Well, looks like the latest offical Appimage 5.1 works! I guess this bug can be closed.

closed 2022-07-23 by GAZ082

chrta, 2022-07-30:
FYI: In fedora 36 (wayland, gnome) i also had to set GDK_BACKEND to be able to launch it (i did not edit /etc/gdm/custom.conf)
```
GDK_BACKEND=x11 LD_PRELOAD=/usr/lib64/libstdc++.so.6 ./Ultimaker-Cura-5.1.0-linux.AppImage
```

fuzebox1, 2022-08-12:
> Well, looks like the latest offical Appimage 5.1 works! I guess this bug can be closed.

Im afraid thar is not the case, at least not for the aur-appimage package.

`(Ultimaker-Cura:9127): GLib-GIO-ERROR **: 16:48:12.495: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'`

This setting is deprecated it seems. Or at least not used in wayland.

For those coming here with the same problem - use _GDK_BACKEND=x11_ cura to run. This can probably be added to the .desktop file or something.

Lets hope for wayland support soon.

joadavis, 2022-12-31:
This item helped me a lot, thanks.

OS: OpenSUSE Leap 15.4
Cura: Ultimaker-Cura-5.2.1-linux.AppImage
command line: `GDK_BACKEND=x11 ./<path>/Ultimaker-Cura-5.2.1-linux.AppImage`

naf623, 2023-03-23:
> 

Try the Linux-Modern version? I just did a fresh install of Ubuntu 22.04 and was suddenly getting this issue, despite not seeing it in my previous install (which had been updated to 22, not a clean install). No problem when I just downloaded the Linux-Modern option.