Ultimaker/Cura #12777: Add graphical printing estimation ala PrusaSlicer.

GAZ082, 2022-07-16:
### Is your feature request related to a problem?

Tweaking to improve the printing time is a must and right now too much time is wasted by hoverig over the small icon and reading the fine print after each slicing process.

### Describe the solution you'd like

PS already solved this by putting a big and intuitive ranking by category of the estimated printing time.

### Describe alternatives you've considered

None.

### Affected users and/or printers

All

### Additional information & file uploads

[image]

Ghostkeeper, 2022-07-27:
We've had to defer this. People agree that it's a nice improvement, quality-of-life. But it doesn't have priority over the other issues currently in our backlog.

We'll see if we get around to adding this later. If there is a redesign of this screen, I'm certain this will be something we'll take into account.

GAZ082, 2023-07-27:
Dont kill this.