Ultimaker/Cura #12778: Bridging orientation issue - v4 & v5 ... a year and still not fixed?

Ltek, 2022-07-16:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Artillery Sidewinder X2

### Reproduction steps

existed in v4.11 (Bridging orientation issue #10372) so opening on for version 5 in hopes someone will fix this

Print a long and thin item that require bridging as in photo...
[image]

### Actual results

Cura set the bridging to happen long-wise (in the photo - its 300mm!) instead of 4mm direction. 

### Expected results

bridge the shortest direction, always

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

project file...
[Cura5_BadBridging.zip](https://github.com/Ultimaker/Cura/files/9126519/Cura5_BadBridging.zip)

smartavionics, 2022-07-16:
Have you tried enabling the bridging settings in the experimental section? If that doesn't do the trick, just try my Cura variant that does what you are expecting. You can find my releases at https://github.com/smartavionics/Cura/releases

[image]

Ltek, 2022-07-16:
Thank you @smartavionics, I'll try that. I didnt know those were there; found with 'Search settings'. I had to turn on 'All' even though 'Expert' was on. 

I still call this a real bug. For example... would anyone ever want to bridge 300mm-- especially since this item has a 100% open end it would be bridging to!  vs the 5mm apart walls? ...  Cura should fix that!

Ghostkeeper, 2022-07-18:
Duplicate of https://github.com/Ultimaker/Cura/issues/10372.