Ultimaker/Cura #12780: Weird shape being generated by Cura 5

luizbgomide, 2022-07-17:
### Application Version

5.0.0 (also happens on beta 5.1.0)

### Platform

Windows 10

### Printer

Two Trees Bluer

### Reproduction steps

Just slice the attached project.

### Actual results

There is a weird loop instead of a flat top. The loop position and format changes depending on orientation and position on the table.
[image]

### Expected results

Smooth top (that gcode was sliced in Cura 4.x).
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[FilamentClipV2.zip](https://github.com/Ultimaker/Cura/files/9127687/FilamentClipV2.zip)

GregValiant, 2022-07-17:
I think this is a duplicate report but I'm unable to find the previous one.  @Ghostkeeper may have information on this behavior.

Ghostkeeper, 2022-07-18:
Yes, it's a duplicate of https://github.com/Ultimaker/Cura/issues/12425.

closed 2022-07-18 by Ghostkeeper