Ultimaker/Cura #12781: неправильная нарезка кура 5 нескольких моделей

Danbaz1983, 2022-07-17:
### Application Version

5,0,0

### Platform

windows10

### Printer

Two trees SP

### Reproduction steps

Загрузил две модели, размножил, нарезал. В модели звезды нарисовано 4 тоннеля. Они печатаются нормально, когда печатается одна звезда.

### Actual results

После размножения модели часть тоннелей исчезла при нарезке
[image]

### Expected results

Тоннели должны быть напечатаны так, как нарисованы.  Любое изменение модели приводит к невозможности нормальной печати, кроме увеличения. Любое масштабирование или изменение приводит к катастрофическим результатам.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[TWOTSPLUS_!Звезда_160_с_вырезом.zip](https://github.com/Ultimaker/Cura/files/9127718/TWOTSPLUS_._160_._.zip)

GregValiant, 2022-07-17:
Please use English for your posts.  I have no way of knowing if I translated your report correctly.

The setting "Make Overhangs Printable" adds chamfers to overhanging areas (like you have within the channels) so that those areas don't require support.  This is expected behavior from that setting.
DIsable "Make Overhangs Printable" and you should get the print you want.  I'll remove the bug label and leave this open so you can let us know if that worked.

Danbaz1983, 2022-07-17:
Я не могу использовать английский, так как это все равно будет машинный перевод. Да, это помогло, спасибо!

closed 2022-07-17 by Danbaz1983

GregValiant, 2022-07-17:
Can this issue be closed?