Ultimaker/Cura #12782: Cura adds strange artifacts

loewexy, 2022-07-17:
### Application Version

5.0.0

### Platform

Ubuntu 20.04

### Printer

Creality Ender 3v2

### Reproduction steps

1. Open the Attached .stl file of a basic box
2. Choose PLA and the standard Quality Profile
3. Slice and look at the preview

### Actual results

The box gets some plastic within.

[image]

### Expected results

The box should print like in the .stl

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[cura_box_problem.zip](https://github.com/Ultimaker/Cura/files/9127792/cura_box_problem.zip)

GregValiant, 2022-07-17:
Before opening your file I'm going to bet I can't reproduce it.
Yep.  I was right.
This problem came up a couple of years ago on the [Ultimaker Forum](https://community.ultimaker.com/topic/36536-cura-generates-geometry-not-present-in-the-original-file/) site.
Here on GitHub #10369 and #11220 appear to be duplicates.
A commonality of the other reports seems to be that the models were exported from SolidWorks.  Is that true in this case?

loewexy, 2022-07-17:
No my model was exported from FreeCAD using the File->Export command. If it is helpful I can also attach the STL.

Ghostkeeper, 2022-07-18:
This is indeed a duplicate of #10369.

Incidentally, it has been fixed for 5.1. The issue occurred with models that have long straight sides. It calculated the area in um² of the triangle that you see in the picture and got an integer overflow there. The algorithm where this occurred has been rewritten in 5.1 and I paid special attention to runaway integer overflows.

closed 2022-07-18 by Ghostkeeper