Ultimaker/Cura #12783: unable to slice in 5.0 but ok in 4.12

gallaghersart, 2022-07-17:
### Application Version

5.0.0

### Platform

windows 10

### Printer

custom coreXY

### Reproduction steps

Like new 5.0 but I can not even do a basic version of my file to work in Cure 5.0 but it works fine in 4.12.
I stripped my FreeCAD design down to its bare minimum and I included the version that works but then once I add one more feature it does not (files that end in “2” are the version that does work in 5.0.).
My final version has many more features (not included in my ZIP) that also do not work but work in 4.12. 

### Actual results

once i press slice a few moments later i get 
"Slicing failed" slicing failed with an unexpected error. please consider reporting.....

### Expected results

good slice

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura_50_20220716.zip](https://github.com/Ultimaker/Cura/files/9127906/cura_50_20220716.zip)
thanks

gallaghersart, 2022-07-17:
So I read though some other people issues ans saw one about not using “connecting infill”. That did not work for me.
So I unchecked “randomize infill start” and it sliced. The basic version I uploaded and my final much more complex model both slice fine

GregValiant, 2022-07-17:
Thanks for the report.
I opened the "Plate Top 2" project file in 5.0 and it sliced fine as it should.
Then I opened it in 5.1beta and there was no problem slicing.  This is on my Windows 10 computer.
I was curious as to why the "Randomiize Infill Start" box was orange (the warning color) and I found this in the Fdmprinter.def.json file:
"warning_value": "True if infill_pattern not in ('grid', 'triangles', 'trihexagon', 'cubic', 'cubicsubdiv', 'tetrahedral', 'quarter_cubic')

So the warning is applied for 7 different types of infill and is not available at all for Lightning infill.  I don't know what the reason was for adding the warning.

Someone from the Cura Team will take a look.  Not being able to duplicate the problem is a problem in itself.

gallaghersart, 2022-07-17:
plate top with out the 2 is the one that does not work. plate top 2 was stripped of a feature to make it work.

GregValiant, 2022-07-17:
Got it.
Using your "Test Bed Level" profile it would not slice.  Using the "Draft" profile it did slice.  I went back to the Test Bed Level profile and played around with settings for a bit.  When I disabled "Randomize Infill Start" the model sliced.

There was another report about that setting interfering with slicing.  I looked at the fdmprinter.def.json file and there are 7 types of infill that will cause the checkbox to turn "warning orange" when it's selected and one of those infill patterns is active.
The infill types that will NOT put up the warning are 'grid', 'triangles', 'trihexagon', 'cubic', 'cubicsubdiv', 'tetrahedral', and 'quarter_cubic'.
The Cura Team will take a look.

Ghostkeeper, 2022-07-18:
We had gotten this issue reported a few times before and got it fixed for the 5.1 release.
I believe the fix is also in the beta. Give it a try!

Both of your project files slice fine in 5.1 stable at least, even though Randomize Infill Start is enabled.

closed 2022-07-18 by Ghostkeeper