Ultimaker/Cura #12784: Show configuration files doesn’t work

Filmman586, 2022-07-17:
### Application Version

5.1.0

### Platform

Windows 10, latest update.

### Printer

Anycubic Vyper

### Reproduction steps

Clicked help, clicked show configuration files.

### Actual results

Nothing. No file location shows up.

### Expected results

I’m assuming file locations

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No files to upload, or should I upload something specific.

fieldOfView, 2022-07-17:
>  should I upload something specific.

Please upload the logs. The cura.log file can be found in `%APPDATA%\cura\5.0` which, incidentally, is the configuration folder that fails to open.

Filmman586, 2022-07-17:
[cura.log](https://github.com/Ultimaker/Cura/files/9128479/cura.log)

Hope it helps, is there another one?

Ghostkeeper, 2022-07-18:
That's the correct file, thank you.

However there is nothing that seems to be going wrong that's logged in that log. Nothing about the `showConfigurationFolder` action. I hoped there would be some QML error about that then, but nothing seems to happen.

I'm not getting the error myself so I can't verify.

One possible cause would be that your Windows username (which I'll not repeat here for possible bots scanning personal info) has a space in it. Perhaps it's actually trying to open `C:\Users\<part before the space>` and then fails on that because that folder doesn't exist. I believe Windows warns you upon creating your Windows user account that it might give trouble with some applications.

MariMakes, 2022-08-24:
Hey @Filmman586 

You can just copy paste %APPDATA%\cura\5.1 in your file explorer bar to navigate to the configuration folder: 
[image]

Can you let me know if it works for you?

closed 2022-09-09 by github-actions[bot]

ki4glx, 2022-09-10:
I had this exact same issue today. I resolved it by using %APPDATA%\cura\5.1. When clicking "Show Configuration Folder" absolutely nothing happens.