Ultimaker/Cura #12786: SLicing Failure

Slart, 2022-07-17:
### Application Version

5.1

### Platform

Win 10

### Printer

Anycubic Vyper

### Reproduction steps

Slicing a file. Tells me there is an error slicing the file.  

Filename is Hex3D_CrackedMjol_V3_P8.stl 

[cura.log](https://github.com/Ultimaker/Cura/files/9128605/cura.log)

### Actual results

tried to slice a file

### Expected results

it should have sliced.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

n/a

Ghostkeeper, 2022-07-18:
How could I reproduce your issue? I tried renaming a cube model file to `Hex3D_CrackedMjol_V3_P8.stl` but it sliced fine.

Could you please share a project file with us? You can create one by going to File -> Save Project... This should allow us to reproduce your issue.

Slart, 2022-07-18:
Copy of the project file attached. Change extension to .3mf
.
[AV_Hex3D_CrackedMjol_V3_P8.txt](https://github.com/Ultimaker/Cura/files/9132737/AV_Hex3D_CrackedMjol_V3_P8.txt)
.

GregValiant, 2022-07-18:
Checking with the Cura Mesh tools indicated the model was watertight but it would not slice.  Repairing the model with MS 3D Builder allowed the model to slice so it did have problems internally.

Here you can see that your infill isn't dense enough.  All those dots are the bottom layer of what comes above and they would all just fall through and the print would fail.
[image]

So this appears to be a model problem and not a Cura problem.

Slart, 2022-07-19:
Interesting. The model slices fine in 5.0 and 4.13, so the issue is 5.1 specific.