Ultimaker/Cura #12788: disappearing details box

mRoNiOnN, 2022-07-18:
### Application Version

5.0.0

### Platform

Mac OS

### Printer

Custom FFF Printer

### Reproduction steps

I do not remember doing anything out of the ordinary. I had just updated Cura to 5.0.0 and noticed this happening.

### Actual results

I hover over a button on a setting that tells me that "this setting is usually calculated", but the box is there for just a fraction of a second before fading away like I dragged the mouse off of it.

### Expected results

The box should remain there until the mouse leaves.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have no more information. Hang in there coders

fieldOfView, 2022-07-18:
I can confirm this issue. Just to clarify the terminology:

The tooltips for the "inherrit button", which shows when the setting has an inherited function, but is overridden by profile, as well as for the "revert button" and the "linked setting icon" all disappear as soon as they are shown, even when the mouse is stil hovering these icons.

mRoNiOnN, 2022-07-18:
> I can confirm this issue. Just to clarify the terminology:
>
> The tooltips for the "inherrit button", which shows when the setting has an inherited function, but is overridden by profile, as well as [...]

Thanks for knowing a lot more than me and fixing it

Ghostkeeper, 2022-07-27:
We added this to our backlog. Developers, see issue CURA-9519.