Ultimaker/Cura #12790: Screen Size Issue

Kwakers01, 2022-07-18:
### Application Version

5 and 4

### Platform

Windows 10

### Printer

Ender3v2

### Reproduction steps

My windows screen size is 1336 x 768 as you can see I cannot see the bits at the bottom of the profile manager.
[image]

### Actual results

As above

### Expected results

Be able to click on the profile manager buttons.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

MariMakes, 2022-08-24:
Hey @Kwakers01,

Thank you for sharing your first issue with us! 🎉 
Sorry it took us a while to get back to you 😢 

I think I recognize your issue!
Can you check what your screen scaling is set to? 
[image]

This is how it looks like for me if scaling is at 150%
[image]

This is how it looks like for me is scaling is at 100%
[image]

Can you let us know if changing your scaling to 100% resolves your issue? 
Thank you! 👼

Kwakers01, 2022-08-24:
Hi, my settings are set to:-

[cid:27b35d83-c04e-4a7e-9b28-330deaf5376a]
I am on an oldish Lenovo Laptop running windows 10.

I checked to see if the settings were compatible and it said in your documentation that the minimum supported was 1280 x 768.

If I upped the scaling, then I would have more of an issue and I do not want to reduce the scaling as I would find the screen difficult to read.

I had a work around in that I pressed (I think it was) ctrl-J and deleted some of the profiles, which is not ideal if you have a lot of them.

Basically the window in your software needs to be fixed to fit the screen size if there is more than a certain number of profiles listed.

Thanks

[186176236-31aa7715-dfbd-42d6-9edc-435059489be2]<https://user-images.githubusercontent.com/40423138/186463333-3c15941a-aea7-4ccd-b494-0960f9f8eefc.png>

[image]<https://user-images.githubusercontent.com/40423138/186463183-801d34ba-bf02-42c1-a340-431b4579da10.png>

[image]<https://user-images.githubusercontent.com/40423138/186463483-92a37067-322f-4bbc-b6e0-3de03e197e9b.png>