Ultimaker/Cura #12793: organise profiles by material

Chukozor, 2022-07-19:
### Is your feature request related to a problem?

It would be a great gain of productivity if we could choose to organize custom profiles by material.
It's really time consuming and not convenient to look for a custom profile you've created plenty of profiles because there are all in the list, not sorted by material..

### Describe the solution you'd like

It would be great if we could at least chose to organize the custom profiles in the exact same way it's done on Ideamaker : when you chose a material "Material A" and then create a profile "MyNewProfile", this profile will be only available when you select "Material A" and won't be available in the list of profile when you have selected other materials like "Material B".
No more big list of profiles mixing up TPU profiles with PETG or PLA ....
It would be a great gain of time and productivity instead of scrolling a lot and looking for the wanted profile in a huge amount of profiles from different material which are all mixed up in the same list.

### Describe alternatives you've considered

maybe a checkbox to chose if the user want to sort the profiles by materials would be great because it will give users the choice to keep it like it is at the moment or to organise the profiles by materials.

### Affected users and/or printers

Everyone

chipmunk1977, 2022-07-20:
I agree that this would be a useful feature.  I'd also like to see the option to link profiles to machines, so that only certain profiles are shown when a particular machine is selected.  currently, I use the workaround of profile names like "material-machine-function" which groups the profiles somewhat.
If I remember rightly, this was discussed some time ago and rejected for design reasons.  Maybe it's time to revisit this?  Alternatively, is this something a plugin could handle?

nallath, 2022-07-20:
>  Alternatively, is this something a plugin could handle?

Kinda. You can store all settings in a material profile with the material settings plugin

>  I'd also like to see the option to link profiles to machines, so that only certain profiles are shown when a particular machine is selected. 

This is again something that is "kinda" possible right now. If you create a profile for the UM3, you will not be able to see that on other printers (except for the Ultimaker 3 Extended). It entirely depends on the choices made by whoever set up the machine profile.

As for having profiles per material, this is a bit more tricky, mostly because dual extrusion machines. We could change this, but this would require a lot of work, which is why it wasn't really picked up.

Chukozor, 2022-07-20:
> Kinda. You can store all settings in a material profile with the material settings plugin

That's not what I'm looking for. As I said, I want to regroup profiles together by materials, and what you suggested is replacing profiles by materials, that's absolutely not the same.

5axes, 2022-07-20:
One intermediat solution : same as for the settings add a filter / search  field on the profile names to simplify the selection of profiles for user with long profile list

Chukozor, 2022-07-20:
indeed, that would help. I can also make several installation in different folders of Cura ..... Gonna do that while waiting/hoping for something to be done about it.

Chukozor, 2022-08-19:
Honestly now I don't care anymore about what's happening in Cura : two weeks ago I've switched to Prusa Slicer and then Super Slicer ; it's mind blowing how good it's thought. So Cura can keep not sorting profiles for all I care ... Still thumbs up for the arachne engine 👍

closed 2022-08-19 by Chukozor