Ultimaker/Cura #12794: Ideas to make it easier to peel off after printing is completed

hiromi3, 2022-07-19:
### Is your feature request related to a problem?

When printing a low-height structure with a 3D printer, there is no place to pinch when peeling it off after printing is completed, which is a problem.

### Describe the solution you'd like

So, for example, if you check "adhesion", it would be convenient to have a function to add a protrusion as shown in the figure around the "adhesion".

### Describe alternatives you've considered

-

### Affected users and/or printers

-

### Additional information & file uploads

[image]

GregValiant, 2022-07-19:
It's not a bad idea.  You can do it with the tools Cura currently provides.
Bring in a couple of support blockers.  Size them the way you want and move then so their brim will merge with the brim of the model.  Under the Per Model settings set them to print as "Normal Model".
As long as the brim was strong enough it might work.  I think it would depend on how strong the brim was connected to the bottom layer of the model and how good the bond was to the build plate.  I have to beat some of prints off the plate with a putty knife and a hammer.
[image]

hiromi3, 2022-07-19:
Thank you for your response.
Since I am learning Cura by myself, there are many functions I do not know.
I selected the function called "Brim" that you taught me (see the figure below).
However, many parameters as shown in your figure are not found in my Cura 5.0.0.
What's wrong?

[image]

fieldOfView, 2022-07-19:
Cura has hundreds of settings. In order not to confuse most users, many of these settings are not shown by default. You can select what settings are visible in the Cura preferences. You can also switch between different sets of visible settings in the "hamburger menu" to the right of the "Search..." box. And finally, you can use that "Search..." box to find settings that are currently not shown by typing their name.

hiromi3, 2022-07-19:
Thank you for your response.
Following your instructions, I was able to display the hidden parameters.
Well, I've found Brim's functionality, but I haven't succeeded in adding the protrusions shown in my first figure.
I found that Cura can specify so many parameters, so I will practice using Cura a little more myself.
Thank you very much.

GregValiant, 2022-07-19:
_"I found that Cura can specify so many parameters..."_  That's true.  It is not a simple piece of software and there is no "Easy Button".  Do as @fieldOfView says and make all the settings visible and play around.  Pay close attention to the preview and slide up and down through the layers to see what is happening during a print and how changing some settings changes the slice.
On the left side toolbar, 6th button down is the Support Blocker button.  Select your model and then click on that button, then select a point on your model and click.  A blocker mesh will be added.  It can be scaled and moved like any other mesh.  In my previous post those little blockers are X5 x Y10 x Z5 and are resting on the build plate at Z=0.

Practice, practice, practice.  Open a gcode file in a text editor and refer to the [MarlinFW ](https://marlinfw.org/meta/gcode/) Site to find out what some of the often used G and M commands do.

3D printing has a steep learning curve but the view from the top of the hill is pretty cool (I just wish I could get up there without a nose bleed).

fieldOfView, 2022-07-19:
> Do as @fieldOfView says and make all the settings visible and play around.

For the record, I DO NOT recommend to make all settings visible. Only make the settings visible that you need regularly, and use the search for the rest. 

Making all settings visible will make the list of settings unnecessarily overwhelming

closed 2022-07-19 by hiromi3