Ultimaker/Cura #12797: [5.0.0] Single layer print is not performed

gexgecko, 2022-07-19:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Create a single layer print (e.g. 0.2mm height with a 0.2mm Z step)
2. Add some pre or post scripts (e.g. draw a purge line, or do a filament change)

### Actual results

The actual print is never performed (only the pre and post scripts)

It's possible that the print has to be small enough (time wise) for this to work, I get a '0 minutes, 0g' estimate in cura after slicing, even though there appears to be valid gcode/extrusion/paths in the preview.

### Expected results

The print should be actually performed.

(This is important for doing first-layer coloured inlays)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_cura-Bug_report.zip](https://github.com/Ultimaker/Cura/files/9141340/CE3_cura-Bug_report.zip)

GregValiant, 2022-07-20:
Thanks for the report.
The first time I tried to open the file using Cura 5.0 it crashed Cura hard.  Then I opened it in 5.1.0 and it sliced.  I went back to 5.0 and it opened and sliced fine.  The Cura Team will take a look.

That model is really thin and delicate.  Have you considered going to a smaller nozzle?
Among your settings you have the Bridge settings and coasting enabled.  Coasting in particular might be a problem with that model,

Asterchades, 2022-07-20:
5.1 is slicing this properly, and the resulting output does include the actual print commands rather than just the start/end codes. I've verified this both manually and using two different GCode viewers. Cura is still showing the 0 estimate but does have an appropriate percentage breakdown, which points to a simple lack of precision being the cause there.

To that end I would suggest updating to 5.1 and trying again, then reporting back if anything has changed (for better or worse).

closed 2022-08-03 by github-actions[bot]