Ultimaker/Cura #12799: Conan install fails

ChrisTerBeke, 2022-07-19:
### Application Version

main

### Platform

MacOS (M1)

### Printer

Not applicable

### Reproduction steps

I ran `conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv` as per the Wiki documentation.

### Actual results

```bash
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    PyQt6-NetworkAuth-Qt6==6.2.4 from https://files.pythonhosted.org/packages/17/0a/4dded3dde322915d532400de2284f5ff20cfa342d60aff8916b6fbf7d852/PyQt6_NetworkAuth_Qt6-6.2.4-py3-none-macosx_11_0_arm64.whl#sha256=1363ea81e5c6ac10bfd643e41ba0d215c0d031a57ff1e5972cc4c2a918efe712 (from -r /Users/chris/.conan/data/uranium/5.1.0-beta+testing_75/ultimaker/testing/package/1208c899a3f6597dd21d1647531d1d5bce281e3f/pip_requirements/requirements.txt (line 22)):
        Expected sha256 1ae9e08e03bd9d5ebdb42dfaccf484a9cc62eeea7504621fe42c005ff1745e66
        Expected     or 8ed4e5e0eaaa42a6f91aba6745eea23fb3ffcbddc6b162016936530ed28dd0ad
        Expected     or c7996a9d8c4ce024529ec37981fbfd525ab1a2d497af1281f81f2b6054452d2e
             Got        1363ea81e5c6ac10bfd643e41ba0d215c0d031a57ff1e5972cc4c2a918efe712

conanfile.py (cura/None): ERROR: Generator VirtualPythonEnv(file:None) failed
Error 1 while executing DYLD_LIBRARY_PATH="" DYLD_FRAMEWORK_PATH="" /Users/chris/Code/Ultimaker/cura/Cura/venv/bin/python -m pip install -r /Users/chris/.conan/data/uranium/5.1.0-beta+testing_75/ultimaker/testing/package/1208c899a3f6597dd21d1647531d1d5bce281e3f/pip_requirements/requirements.txt --force-reinstall
ERROR: Error 1 while executing DYLD_LIBRARY_PATH="" DYLD_FRAMEWORK_PATH="" /Users/chris/Code/Ultimaker/cura/Cura/venv/bin/python -m pip install -r /Users/chris/.conan/data/uranium/5.1.0-beta+testing_75/ultimaker/testing/package/1208c899a3f6597dd21d1647531d1d5bce281e3f/pip_requirements/requirements.txt --force-reinstall
```

### Expected results

The installation process succeeds and I can run `python cura_app.py` afterwards.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

ChrisTerBeke, 2022-07-28:
Unfortunately changing both occurrences of `armv8` to `x86_64` in `.conan/profile/default` did not work, another error popped up:

```
ld: warning: ignoring file Programs/python.o, building for macOS-arm64 but attempting to link with file built for unknown-x86_64
ld: warning: ignoring file Programs/_testembed.o, building for macOS-arm64 but attempting to link with file built for unknown-x86_64
ld: warning: ignoring file /Users/chris/.conan/data/libxcrypt/4.4.25/_/_/package/299aaabcb5fb1385c096ccf2c550a653c4438173/lib/libcrypt.a, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /Users/chris/.conan/data/libxcrypt/4.4.25/_/_/package/299aaabcb5fb1385c096ccf2c550a653c4438173/lib/libcrypt.a, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "_main", referenced from:
     implicit entry/start for main executable
Undefined symbols for architecture arm64:
ld: symbol(s) not found for architecture arm64
  "_main", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Programs/_testembed] Error 1
make: *** Waiting for unfinished jobs....
make: *** [python.exe] Error 1
cpython/3.10.4: 
cpython/3.10.4: ERROR: Package '14d68bd42e0ec1c94a90ce04b8da2364cced4342' build failed
cpython/3.10.4: WARN: Build folder /Users/chris/.conan/data/cpython/3.10.4/_/_/build/14d68bd42e0ec1c94a90ce04b8da2364cced4342
ERROR: cpython/3.10.4: Error in build() method, line 428
	autotools.make()
	ConanException: Error 2 while executing make -j10
```

MariMakes, 2022-08-24:
Hey!!! @ChrisTerBeke 

Good to see you here 😄 

We've updated the Running from Source wiki a few days ago. 
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

Can you check if that helps you along? 
If not... Can you tag @Joeydelarago so he can help you along?

jellespijker, 2022-08-24:
And @ChrisTerBeke please make sure you also clean the Conan cache, just to be sure that you're not mixing up different architectures.

```
conan remove "*" -f
```

ChrisTerBeke, 2022-08-25:
Hi! The updated instructions work a lot better, especially using a Rosetta Terminal (maybe add a link to a guide on how to use Rosetta in a Terminal session? I had to Google for it ;) ).

However I now run into an issue with building `libnest2d`:

```
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: Configuring sources in /Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/source/.
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: SCM: Getting sources from url: 'https://github.com/Ultimaker/libnest2d'
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: Copying sources to build folder
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: Building your package in /Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: Generator txt created conanbuildinfo.txt
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: Calling generate()
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: Aggregating env generators
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: Calling build()
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: CMake command: cmake -G "Ninja" -DCMAKE_TOOLCHAIN_FILE="/Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/build/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="/Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/package/424b75ce0ac32e98316391eb44b75be421094393" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Release" "/Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/."
Capturing current environment in /Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/build/generators/deactivate_conanbuildenv-release-armv8.sh
Configuring environment variables
-- Using Conan toolchain: /Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/build/generators/conan_toolchain.cmake
-- Conan toolchain: C++ Standard 20 with extensions OFF
-- Conan toolchain: Setting BUILD_SHARED_LIBS = ON
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting warnings for project_options
-- Conan: Target declared 'clipper::clipper'
-- Conan: Component target declared 'Boost::headers'
-- Conan: Component target declared 'Boost::boost'
-- Conan: Target declared 'boost::boost'
-- Conan: Component target declared 'NLopt::nlopt'
-- Enabling threading support for project_options
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_POLICY_DEFAULT_CMP0091

-- Build files have been written to: /Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/build/Release
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: CMake command: cmake --build "/Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/build/Release" '--' '-j10'
Capturing current environment in /Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/build/generators/deactivate_conanbuildenv-release-armv8.sh
Configuring environment variables
[1/2] Building CXX object CMakeFiles/nest2d.dir/src/libnest2d.cpp.o
FAILED: CMakeFiles/nest2d.dir/src/libnest2d.cpp.o 
/Library/Developer/CommandLineTools/usr/bin/c++ -DLIBNEST2D_GEOMETRIES_clipper -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_THREADING_std -Dnest2d_EXPORTS -I/Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/include -I/Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/src -isystem /Users/chris/.conan/data/clipper/6.4.2/_/_/package/2e4ccf36d410220ebd34951aaeaabe88b32b5bcb/include -isystem /Users/chris/.conan/data/boost/1.79.0/_/_/package/524ea35a8120baabdde02483add58d81bf541327/include -isystem /Users/chris/.conan/data/nlopt/2.7.0/_/_/package/a93bbaf727996ad606a19d0b172803109bf585a5/include -stdlib=libc++ -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -fPIC -Wall -Wextra -Wshadow -Wnon-virtual-dtor -Wold-style-cast -Wcast-align -Wunused -Woverloaded-virtual -Wpedantic -Wconversion -Wsign-conversion -Wnull-dereference -Wdouble-promotion -Wformat=2 -std=c++20 -MD -MT CMakeFiles/nest2d.dir/src/libnest2d.cpp.o -MF CMakeFiles/nest2d.dir/src/libnest2d.cpp.o.d -o CMakeFiles/nest2d.dir/src/libnest2d.cpp.o -c /Users/chris/.conan/data/nest2d/5.2.0-alph [... cut ...]
nest2d/5.2.0-alpha+testing_26@ultimaker/testing: WARN: Build folder /Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/build/Release
ERROR: nest2d/5.2.0-alpha+testing_26@ultimaker/testing: Error in build() method, line 110
	cmake.build()
	ConanException: Error 1 while executing cmake --build "/Users/chris/.conan/data/nest2d/5.2.0-alpha+testing_26/ultimaker/testing/build/424b75ce0ac32e98316391eb44b75be421094393/build/Release" '--' '-j10'
```

ChrisTerBeke, 2022-08-25:
Something else to note is that I had to pass a setting to use CPP 20 (instead of the default CPP 17 on my machine) using a build setting:

```
conan install . --build=missing --update -o cura:devtools=True -s compiler.cppstd=20 -g VirtualPythonEnv
```

jellespijker, 2022-08-25:
Only the engine is compiled with c++20. Libnest2d isn't compatible with the C++20 standard. It should be defined in the profile to only compile Curaengine with c++20

Did you update the Conan configuration? I think you might be working with an old configuration

Otherwise try using `-s curaengine:compiler.cppstd=c++20`

jellespijker, 2022-08-25:
> The updated instructions work a lot better, especially using a Rosetta Terminal (maybe add a link to a guide on how to use Rosetta in a Terminal session? I had to Google for it ;) ).

@Joeydelarago good work on the instructions.
Can you think about the value of adding the Rosetta comment?

ChrisTerBeke, 2022-08-26:
Thanks! Setting up a new conan profile worked.

Now another error occurred when building `pyarcus`:

```
pyarcus/5.2.0-alpha+testing_27@ultimaker/testing: Configuring sources in /Users/chris/.conan/data/pyarcus/5.2.0-alpha+testing_27/ultimaker/testing/source/.
pyarcus/5.2.0-alpha+testing_27@ultimaker/testing: SCM: Getting sources from url: 'https://github.com/Ultimaker/pyArcus'
pyarcus/5.2.0-alpha+testing_27@ultimaker/testing: Copying sources to build folder
pyarcus/5.2.0-alpha+testing_27@ultimaker/testing: Building your package in /Users/chris/.conan/data/pyarcus/5.2.0-alpha+testing_27/ultimaker/testing/build/55eabcc31f4737a5d42eb88eb3c5a43a5361d7d9
pyarcus/5.2.0-alpha+testing_27@ultimaker/testing: Generator txt created conanbuildinfo.txt
pyarcus/5.2.0-alpha+testing_27@ultimaker/testing: Calling generate()
['\n[build-system]\nrequires = ["sip >=6, <7", "setuptools>=40.8.0", "wheel"]\nbuild-backend = "sipbuild.api"', '\n[tool.sip.metadata]\nname = "pyArcus"\nversion = "5.2.0a"\nsummary = "Communication library between internal components for Ultimaker software"\nhome-page = "https://github.com/Ultimaker/libArcus"\nauthor = "Ultimaker B.V."\nlicense = "LGPL-3.0"\ndescription-file = "README.md"\nrequires-python = ">=3.10.4"', '\n[tool.sip.project]\ncompile = false\nsip-files-dir = "python"\nbuild-dir = "/Users/chris/.conan/data/pyarcus/5.2.0-alpha+testing_27/ultimaker/testing/build/55eabcc31f4737a5d42eb88eb3c5a43a5361d7d9/build/Release"\ntarget-dir = "/Users/chris/.conan/data/pyarcus/5.2.0-alpha+testing_27/ultimaker/testing/package/55eabcc31f4737a5d42eb88eb3c5a43a5361d7d9/site-packages"\npy-include-dir = "/Users/chris/.conan/data/cpython/3.10.4/_/_/package/eb313874fa95471108a4cf3448d19f6d390c0a3c/include/python3.10"\npy-major-version = 3\npy-minor-version = 10', "\n[tool.sip.bindings.pyArcus]\nexceptions = true\nrelease-gil = true\nlibraries = ['Arcus', 'python3.10', 'protoc', 'protobuf', 'ssl', 'crypto', 'expat', 'ffi', 'mpdec++', 'mpdec', 'crypt', 'bz2', 'gdbm', 'sqlite3', 'tk8.6', 'tkstub8.6', 'panelw', 'menuw', 'formw', 'ncurses++w', 'ticw', 'ncursesw', 'tinfow', 'lzma', 'tdbcmysql1.1.1', 'itcl4.2.0', 'itclstub4.2.0', 'tdbcodbc1.1.1', 'tdbcpostgres1.1.1', 'tdbc1.1.1', 'tdbcstub1.1.1', 'thread2.8.5', 'sqlite3.30.1.2', 'tcl8.6', 'tclstub8.6', 'z']\nlibrary-dirs = ['/Users/chris/.conan/data/arcus/5.2.0-alpha+testing_23/ultimaker/testing/package/5193ec95fede3e06794d0aaf53918122cac23535/lib', '/Users/chris/.conan/data/cpython/3.10.4/_/_/package/eb313874fa95471108a4cf3448d19f6d390c0a3c/lib', '/Users/chris/.conan/data/protobuf/3.21.4/_/_/package/22ab557dfe77599e69da6a6d2ce786fe3e213850/lib', '/Users/chris/.conan/data/openssl/1.1.1l/_/_/package/095512ed878f14a09dd732e9f6868729dd437529/lib', '/Users/chris/.conan/data/expat/2.4.1/_/_/package/38ab6a38f66c07eba71698df5ea91b7fed6840bf/lib', '/Users/chris/.conan/data/libffi/3.2.1/_/_/package/095512ed878f14a09dd732e9f6868729dd437529/lib', '/Users/chris/.conan/data/mpdecimal/2.5.0/_/_/package/3803a313e9f749c39d88c658650fa5f27b6129ce/lib', '/Users/chris/.conan/data/libxcrypt/4.4.25/_/_/package/e98cc99c78f8ae352930d807a85fc7d6df891023/lib', '/Users/chris/.conan/data/bzip2/1.0.8/_/_/package/2b8e4ed96946ad0a4cd2c6e1e01636aa2a421589/lib', '/Users/chris/.conan/data/gdbm/1.19/_/_/package/321f4842541d720fe1e44247f6661ffba9121a53/lib', '/Users/chris/.conan/data/sqlite3/3.36.0/_/_/package/509c6f4636dec8645bfd459c6049cbef21b8ce1b/lib', '/Users/chris/.conan/data/tk/8.6.10/_/_/package/c8df73d0837278969de8aa4036da5b007cc60373/lib', '/Users/chris/.conan/data/ncurses/6.2/_/_/package/c50d83884369def95bce6d55de9b1884aa3b2950/lib', '/Users/chris/.conan/data/xz_utils/5.2.5/_/_/package/095512ed878f14a09dd732e9f6868729dd437529/lib', '/Users/chris/.conan/data/tcl/8.6.10/_/_/package/385f29aec40a23393eafbfe8d365b2892ed5361c/lib/tdbcmysql1.1.1', '/Users/chris/.conan/data/tcl/8.6.10/_/_/package/385f29aec40a23393eafbfe8d365b2892ed5361c/lib/itcl4.2.0', '/Users/chris/.conan/data/tcl/8.6.10/_/_/package/385f29aec40a23393eafbfe8d365b2892ed5361c/lib/tdbcodbc1.1.1', '/Users/chris/.conan/data/tcl/8.6.10/_/_/package/385f29aec40a23393eafbfe8d365b2892ed5361c/lib/tdbcpostgres1.1.1', '/Users/chris/.conan/data/tcl/8.6.10/_/_/package/385f29aec40a23393eafbfe8d365b2892ed5361c/lib/tdbc1.1.1', '/Users/chris/.conan/data/tcl/8.6.10/_/_/package/385f29aec40a23393eafbfe8d365b2892ed5361c/lib/thread2.8.5', '/Users/chris/.conan/data/tcl/8.6.10/_/_/package/385f29aec40a23393eafbfe8d365b2892ed5361c/lib/sqlite3.30.1.2', '/Users/chris/.conan/data/tcl/8.6.10/_/_/package/385f29aec40a23393eafbfe8d365b2892ed5361c/lib', '/Users/chris/.conan/data/zlib/1.2.12/_/_/package/095512ed878f14a09dd732e9f6868729dd437529/lib']\ninclude-dirs = ['/Users/chris/.conan/data/arcus/5.2.0-alpha+testing_23/ultimaker/testing/package/5193ec95fede3e06794d0aaf53918122cac23535/include', '/Users/chris/.conan/data/cpython/3.10.4/_/_/package/eb313874fa95471108a4cf3448d19f6d390c0a3c/include', '/Users/chris/.conan/data/cpython/3.10.4/_/_/package/eb313874fa95471108a4cf3448d19f6d390c0a3c/include/python3.10', '/Users/chris/.conan/data/protobuf/3.21.4/_/_/package/22ab557dfe77599e69da6a6d2ce786fe3e213850/include', '/Users/chris/.conan/data/openssl/1.1.1l/_/_/package/095512ed878f14a09dd732e9f6868729dd437529/include', '/Users/chris/.conan/data/expat/2.4.1/_/_/package/38ab6a38f66c07eba71698df5ea91b7fed6840bf/include', '/Users/chris/.conan/data/libffi/3.2.1/_/_/package/095512ed878f14a09dd732e9f6868729dd437529/include', '/Users/chris/.conan/data/mpdecimal/2.5.0/_/_ [... cut ...] honMessage.cpp']", "\nextra-compile-args = ['-stdlib=libc++', '-std=c++17', '-fPIC', '-O3']\nextra-link-args = []"]
pyarcus/5.2.0-alpha+testing_27@ultimaker/testing: Calling:
 > "sip-build"

----Running------
> "sip-build"
-----------------
/bin/sh: sip-build: command not found
ERROR: pyarcus/5.2.0-alpha+testing_27@ultimaker/testing: Error in generate() method, line 96
	sip.build()
while calling 'build', line 34
	self._conanfile.run(cmd)
	ConanException: Error 127 while executing "sip-build"
```

ChrisTerBeke, 2022-08-26:
> Did you update the Conan configuration? I think you might be working with an old configuration

This was indeed the issue, I can run Cura from source now!

I had to run these lines again to refresh the configuration:

```
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
```

I'm not very familiar with Conan, so this was not something that came to my mind without your hint. I thought pulling the latest Cura main branch would solve things :P

Thanks!

closed 2022-08-26 by ChrisTerBeke

jellespijker, 2022-08-26:
That is greet news @ChrisTerBeke 

That is why one of the changes to the config, contains an hourly update check, with an automatic pull (lazy execution so no background process running). So now on out you don't have to think about that anymore.
I might lower that frequency once things are more stable.

julianCast, 2022-11-08:
I'm facing the exact same initial issue but using an Intel Mac (no M1). I did refresh the configuration, but same issue with `Error 1 while executing DYLD_LIBRARY_PATH`