Ultimaker/Cura #12800: Seam line

Elboyishere, 2022-07-19:
### Application Version

5.0.0

### Platform

Windows11

### Printer

ender 5 plus

### Reproduction steps

I was using 4.1 and after the update, I noticed my printing was off. I did some adjustments to my settings and now I noticed a seam. I played with the settings and something just not right. 

### Actual results

Seam just gets spoty in the back

### Expected results

I never had an issue with the seam with 4.1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Calibration tower also needs something it stops for a second before going to other side. My nozzle gets clogged when I run the test.

Elboyishere, 2022-07-22:
> Hi @Elboyishere could you please share a project file with us so we can try and reproduce your issue? Thank you!

Seam with smart hiding

Elboyishere, 2022-07-22:
Printed with 4.1 software

fieldOfView, 2022-07-22:
Unfortunately github scrubs attachments sent via email replies. You are going to have to add any attachments via the webinterface here: https://github.com/Ultimaker/Cura/issues/12800

tlhintoq, 2022-07-22:
[image]

Personal guess... Seam placement insists on disabling the option to even choose "none".  Its insists on trying to place at "back".
Which I suspect it looks for where the *back* is in real time while slicing and comes up with a different number on every layer, so its all over the place.

In 4.13 we could actually set "none".  In 5.0/5.1 we can't.
[image]

Elboyishere, 2022-07-22:
Makes sense. I tried different options they give on settings but still has seam…. I was going to try and load the old software and compare the prints side by side….

Vandresc, 2022-07-25:
Hi @Elboyishere , could be me but I still don't see a project file attached. Could you pls see fieldsofview comment again and retry?Thank you!

tlhintoq, 2022-08-08:
ANy of us would have attached a project; not just the original author.  If you re-open I'll give you the one used when generating the screenshots I provided.

tlhintoq, 2022-08-08:
[CoffeeV4.zip](https://github.com/Ultimaker/Cura/files/9281829/CoffeeV4.zip)
Attached coffee cup project used for the screen shots above.
As you can see in those shots, and with your use of it... On 4.13 you get a great straight seem.  With 5.x is all over the place.

tlhintoq, 2022-08-08:
@Vandresc 
And of course the bot closed it again

Vandresc, 2022-08-08:
Hi @tlhintoq, I am not getting the same results as you do. Left is 5.1; Right is 4.13. But also, I used one of our printers and a creality printer for this, as your machine definition is unknown so it will only load the stl. We do however have similar tickets on our backlog, so I will add this case to our tickets as well. 

[image]

Vandresc, 2022-08-08:
Duplicate of https://github.com/Ultimaker/Cura/issues/12355, ticket on the backlog

tlhintoq, 2022-08-08:
That is a terrific 5.1 seam.   Can you share a screenshot of your seam values?
Which machine did you configure this for?  I'll change my project to the same machine and see what I get.

Vandresc, 2022-08-09:
Values I used where the default User specified seam for a Creality ender 5 (seam X 110; seam Y 220) and an Ultimaker S5 (seam X 165; seam Y 240), a bit of different result when comparing the two printers but expected as they don't share the same quality profiles. Ticket is on our backlog.

tlhintoq, 2022-08-09:
Sorry if I sound like a schmuck but... You guys like complete information, right?  It helps us reproduce _your_ results if you do the same.  There's a lot more than xy values for seams and that's why we give you screen shots in the hopes of tracking down all the permutations like seam location and [x] Relative etc.  It would help us if you could do that in return.

There isn't much more of a generic machine than the Ender3 - everybody has them.  Everybody gets them as their first machine.   So I've attached an updated project with the same cup on an Ender3 - but I get the same results on Hictop D3Hero, and CR10s-5, and Anycubic KobraMax.

Here's that same cup on an Ender3pro under 4.13 and 5.1 - Same settings.  And like everyone else on 3 tickets for this same issue: It comes out right on 4.13 and bad on 5.1 - I don't know what to tell ya.
[image]

One observation I did make based on your values is the seam location MIGHT need to be WAY outside the model.  I guess there's some algorithm for mapping that back onto the model.  But if the seam location is _INSIDE_ the model I think there's a higher likely hood of a bad seam.  Kind of like it can find a clean position if it has to shrink in toward the model, but its rough if it has to expand outward.
150x220 and shrink inward good
10x20 and expand outward rough
[2022-08-09_08-32-19.zip](https://github.com/Ultimaker/Cura/files/9290542/2022-08-09_08-32-19.zip)

Vandresc, 2022-08-09:
No, not in this case, it's fair! ;) 
I am getting inconsistent results now though; because my relative seam was off and I switched between printers that have different build plates, the model got into a position where my seam looked "smooth", I think it was just a lucky shot. I attached a project where I could find a more optimal seem on your model but as soon as the relative seam gets locked then the seam gets random again. 

[image]

As we already know something is really off with the seam we will really strive to fix some of the seam issues soon. We will update the git tickets once our internal tickets are making progress. Again, thank you for sharing the projects and the extra information, this should help analyze the ticket a bit faster. 
[cappuccino.zip](https://github.com/Ultimaker/Cura/files/9291588/cappuccino.zip)

tlhintoq, 2022-08-09:
> I am getting inconsistent results now though

It is SO very comforting to know you were able to reproduce this!  As a developer myself (Visual Studio, cross-platform stuff), I know that very often just reproducing it is 90% of the battle.  I often say "I can't fix what I can't break".

Glad to have been able to help steer in a direction to investigate.

GregValiant, 2026-08-19:
It's been 4 years so I'll go ahead and close this.

closed 2026-08-19 by GregValiant