Ultimaker/Cura #12802: Layer View Visual Bug/Glitch

ChaosBlades, 2022-07-19:
### Application Version

5.1.0

### Platform

Pop!_OS 22.04 LTS x86_64

### Printer

Voron 2 350

### Reproduction steps

1. Launch Cura with clean config
2. Slice any object (3MF for me)
3. Open Layer View Preview and rotate the camera

On the very first slice once you enter preview mode the model will not be glitched but as soon as you rotate the camera the model glitches out. After this first glitch occures it will always be there even if you reslice or replace the model with another.

### Actual results

Layer View has a visual Bug/Glitch

[image]

### Expected results

No visual bugs in layer view

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Log File:
[cura.log](https://github.com/Ultimaker/Cura/files/9144747/cura.log)
 
Project File:
[Visual Bug.zip](https://github.com/Ultimaker/Cura/files/9144757/Visual.Bug.zip)

I have two system. My laptop does not have this issue but my desktop does. I thought this might have something to do with my 21:9 monitor but setting it to 1080p and relaunching Cura did not resolve the issue. They are both all AMD systems running the same OS and both are up to date.

Both systems are running `Ultimaker-Cura-5.1.0-linux-modern.AppImage` the .config folder was deleted before testing.

I tested all the way back to version 4.9.0 and it still has this issue.

Laptop (No Glitch)
CPU: AMD Ryzen 7 4800H
RAM: 16GB
GPU: Integrated
Resolution: 1920x1080

Desktop (Glitch)
CPU: AMD Ryzen 9 5950X
RAM: 32GB
GPU: AMD Radeon RX 6900XT
Resolution: 3440x1440

jellespijker, 2022-07-25:
Thank you for bringing this up. My first thought was that you had `compatibility mode` on. But looking more closely at the screenshots it looks like it isn't drawing all the faces for each line.

I will bring this up with the team.

Vandresc, 2022-07-26:
Hi @ChaosBlades , we are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.
To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
Thank you for bringing this up to our attention.

nnmalex, 2022-08-02:
It's worth mentioning that this was introduced in 5.1 and 5.0 worked correctly.

warpKaiba, 2022-08-28:
I have this same issue. Here's my info in case it helps anyone in the future.
It only effects layers that aren't currently the top-most visible layer. Here's what it looks like up close:

Layers 1-2:
[image]

Just Layer 1:
[image]

My specs:
OS: Manjaro 21.3.7 Ruah
Kernel: x86_64 Linux 5.15.60-1-MANJARO
Resolution: 5120x2520
DE: KDE 5.96.0 / Plasma 5.24.6
WM: KWin
CPU: AMD Ryzen 9 5950X 16-Core @ 32x 3.4GHz
GPU: AMD Radeon RX 6800 (sienna_cichlid, LLVM 14.0.6, DRM 3.42, 5.15.60-1-MANJARO)

nfp0, 2023-02-26:
I also have a Radeon RX 6800 card and I'm on Manjaro.
It's seems all users affected by this bug have Radeon RX 6000 series cards and are on Linux. Seems like a pattern.

Normal image:
[image]

Glitchy:
[image]

Sh4dow, 2023-02-28:
I have the same problem and also AMD system. Radeon 6700xt and Ryzen 3600 under Kubuntu. 
Cura on the same machine but under windows is working. So maybe it is drivers issue?

```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: AMD (0x1002)
    Device: AMD Radeon RX 6700 XT (navy_flounder, LLVM 14.0.1, DRM 3.47, 5.19.0-31-generic) (0x73df)
    Version: 22.1.0
    Accelerated: yes
    Video memory: 12288MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 9783 MB, largest block: 9783 MB
    VBO free aux. memory - total: 15836 MB, largest block: 15836 MB
    Texture free memory - total: 9783 MB, largest block: 9783 MB
    Texture free aux. memory - total: 15836 MB, largest block: 15836 MB
    Renderbuffer free memory - total: 9783 MB, largest block: 9783 MB
    Renderbuffer free aux. memory - total: 15836 MB, largest block: 15836 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 12288 MB
    Total available memory: 28289 MB
    Currently available dedicated video memory: 9783 MB
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon RX 6700 XT (navy_flounder, LLVM 14.0.1, DRM 3.47, 5.19.0-31-generic)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.1.0-devel
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.1.0-devel
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.1.0-devel
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

ChaosBlades, 2023-02-28:
I actually use SuperSlicer even back when I first posted this so I have not been on top of this issue and have not tried to launch Cura in a very long time.

Just downloaded the latest version of Cura 5.2.2 off the website and it will not even launch for me. Looks like I am suffering from https://github.com/Ultimaker/Cura/issues/11856. I had to download the "linux-modern" release from GitHub which is also 5.2.2.

With that version I am not seeing any visual glitch other than Cura STILL not having any anti-aliasing https://github.com/Ultimaker/Cura/issues/7447.

I am still running Pop_OS 22.04 LTS, 5950X, and 6900XT but now on kernel 6.1.11-76060111-generic. I am also running Mesa 23.1.0-devel which is what I suspect is fixing the issue. Can someone else try to use the latest development version of Mesa to see if that fixes the issue?

ulmerkott, 2023-04-23:
> I am still running Pop_OS 22.04 LTS, 5950X, and 6900XT but now on kernel 6.1.11-76060111-generic. I am also running Mesa 23.1.0-devel which is what I suspect is fixing the issue. Can someone else try to use the latest development version of Mesa to see if that fixes the issue?

Uppgrading to Mesa 23.2.0-devel (the current bleeding edge) fixes the rendering issue for me.

My specs:
Ubuntu 22.04
CPU: AMD Ryzen 9 6900HX
GPU: Radeon 680M (Integrated)