Ultimaker/Cura #12803: Cura 5.1.0 appimage Failed to probe OpenGL however 5.0.0 appimage still works

nathanshearer, 2022-07-19:
### Application Version

5.1.0

### Platform

Linux

### Printer

Voron 2.4 350

### Reproduction steps

1. Use Cura 5.0.0 without problems
2. Open Cura 5.1.0 and receive error "Failed to probe OpenGL"

### Actual results

[image]

### Expected results

The Cura 5.0.0 appimage works without problems. I expect the 5.1.0 appimage to also work too.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

My system has no problems launching the appimage for Cura 4.10.0 through 5.0.0 however 5.1.0 will not launch.

nathanshearer, 2022-07-19:
[glxinfo.txt](https://github.com/Ultimaker/Cura/files/9145058/glxinfo.txt)

jellespijker, 2022-07-22:
Looks like a duplicate of #11856 which is fixed in 5.1.0 by providing an AppImage based on a modern 2022 distro.

Can you tell me which AppImage you used and what the output is when you run:
```
ldd --version 
```

nathanshearer, 2022-07-22:
I downloaded this release: https://github.com/Ultimaker/Cura/releases/download/5.1.0/Ultimaker-Cura-5.1.0-linux.AppImage
```
$ ldd --version
ldd (Gentoo 2.34-r10 p15) 2.34
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

gaima8, 2022-07-24:
Same here.

```
$ ldd --version
ldd (Gentoo 2.34-r13 p17) 2.34
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```
```
$ ./Downloads/Ultimaker-Cura-5.1.0-linux.AppImage
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_UltimagqXJg3/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib64/dri/iris_dri.so) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /tmp/.mount_UltimagqXJg3/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib64/dri/iris_dri.so) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_UltimagqXJg3/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib64/dri/swrast_dri.so) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: swrast
Fontconfig warning: "/etc/fonts/conf.avail/05-reset-dirs-sample.conf", line 6: unknown element "reset-dirs"
```

The libstdc++.so.6 within Ultimaker-Cura-5.0.0-linux.AppImage has the symbol GLIBCXX_3.4.29, the libstdc++.so.6 in Ultimaker-Cura-5.1.0-linux.AppImage only goes up to GLIBCXX_3.4.28.

I've not been able to find where, or how, the release packages are built, but it seems like a regression on the "not modern" build.

jellespijker, 2022-07-24:
Try the modern build, that build was specifically released for systems with newer glibc versions

closed 2022-07-24 by jellespijker