Ultimaker/Cura #12804: Cura 5.1.0, Win11, Support showing up in preview when it should not

csterlin, 2022-07-20:
### Application Version

5.1.0

### Platform

Windows 11, 11th Gen Intel(R) Core(TM) i7-11390H, 16.0 GB, Intel IRIS Xe

### Printer

Creality Ender 3

### Reproduction steps

1. Sliced using 0.20mm layer height, using support touching buildplate.
2. Sliced using 0.16mm layer height, using support touching buildplate.

### Actual results

1. Support using 0.20mm layer height, shows in preview as set to everywhere.
2. Support using 0.16mm layer height, shows in preview as set to touching buildplate.

### Expected results

1. Support should show as set to touching buildplate, not as eveywhere. This is incorrect behavior.
2. Support shows as set, to touching buildplate. This is the expected, correct behavior.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Tower1_0.20LayerHeight.3mf.zip](https://github.com/Ultimaker/Cura/files/9146708/CE3_Tower1_0.20LayerHeight.3mf.zip)

[image]
[image]
[CE3_Tower1_0.16LayerHeight.3mf.zip](https://github.com/Ultimaker/Cura/files/9146716/CE3_Tower1_0.16LayerHeight.3mf.zip)

I have not tried to actually print this object using either layer height, so I do not know if the anomalous behavior shows up in the finished product. I am trying to print it as I type this note.

csterlin, 2022-07-20:
Finished print of the 0.20mm layer height object, and it demonstrates the anomalous behavior.
[image]

csterlin, 2022-07-21:
Here is another example of bad application of supports
[image]
.

Vandresc, 2022-07-22:
Hi @csterlin I will bring it up with the rest of the team. Thank you for your feedback.

Vandresc, 2022-07-26:
Hi @csterlin, we believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

We do have some workarounds for you that will "fix" your issue:
1. Disable conical support from experimental, this will remove the support from the top part of the model, or
2. Adjust the Support XY Distance to remove or make the support altogether connect.

Thank you again for your feedback and the clear description of your issue.

csterlin, 2022-07-26:
Thank you for your helpful response. I had not considered conical support
contributing to the issue, even though it remains experimental. Again,
thanks.