Ultimaker/Cura #12805: Cura 5.1: Slicing failed with an unexpected error

neutebach, 2022-07-20:
### Application Version

5.1

### Platform

Windows 10

### Printer

E3D Toolchanger (fdmprinter)

### Reproduction steps

fresh install of Cura 5.1, although old profiles and settings have been pulled into the new version. Opened an older project (might have been from arachne beta 2 or Cura 5.0), replaced the STL model, tried to slice, unknown error, Cura asks me to report a bug so here we go. see log attached.

### Actual results

[image]

### Expected results

[image]
(same project works well in Cura 5.0)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[project + log.zip](https://github.com/Ultimaker/Cura/files/9147984/project.%2B.log.zip)

GregValiant, 2022-07-21:
Thanks for the report.
It sliced for me in 5.1.0.  Someone from the Cura Team will take a look.

You might want to go over your settings.  The skirt extruder is extruder 2 and that's all it is doing.  You could switch the skirt to extruder 1 and then turn off the prime tower.  That won't keep it from slicing though.

neutebach, 2022-07-21:
thank you Greg. Interesting that it worked for you.
I skimmed through the log shortly and found some remarks as to why the slicing engine stopped, but couldn't make much of it, someone more knowledgeable will surely find out why it didn't work for me.

I'm aware of the settings issues, please disregard, since Cura doesn't handle multi-tool setups very well, I have to go over many settings each time I change - for example - filaments... but that's another story.

Ghostkeeper, 2022-07-21:
It seems that this entry from the log indicates what the issue is:

> [ERROR] Trying to retrieve setting with no value given: 'acceleration_travel_enabled'

This means that CuraEngine is new, but the front-end (or at least the settings list) is outdated somehow.

Maybe this could happen if you installed 5.1 on top of the same folder in Program Files as 5.0? Or perhaps you have a copy of fdmprinter in your configuration folder? (In Cura go to Help -> Show Configuration Folder, and then search using the file browser for files named `fdmprinter.def.json`.)
It does not reproduce in any of the builds by default as far as I could test it.

closed 2022-08-05 by github-actions[bot]