Ultimaker/Cura #12807: Two times infill in same layer.

Havukaski, 2022-07-20:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Custom printer

### Reproduction steps

Sliced object made in SolidWorks 2021. Printer is large scale granule printer.

While printing noticed that printer tried to do infill twice for same layer, which was confirmed by gcode viewer to be problem in gcode.

Attached: Project file and gcode.

### Actual results

Both layers 5 and 7 have 2 identical infills in top of each other i.e. it is trying to print same area twice.

### Expected results

Normal printing

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[KulmaLaatikko2.zip](https://github.com/Ultimaker/Cura/files/9149357/KulmaLaatikko2.zip)

GregValiant, 2022-07-20:
Thanks for the report.
 I can duplicate the problem using your project file.  This behavior has come up before in #11401, and #11537.
I'll mark it as a duplicate.  Someone from the Cura Team will take a look.  The same problem occurs with 5.1.0.

These lines are all in gcode Layer:4:
G1 X-47.999 Y-62.141 E2053.75132
G1 X-47.999 Y-59.313 E2056.57311

G1 X-47.999 Y-62.141 E3709.94768
G1 X-47.999 Y-59.313 E3712.76947

Vandresc, 2022-07-26:
Devs see CURA-9515 for our internal reference. I will also add the scenarios from the duplicated issues that @GregValiant mentioned, though that issue had a workaround, and I still get the double layer when the pattern is Lines and not ZigZag. Thank you both!

nallath, 2022-09-02:
This was fixed by https://github.com/Ultimaker/CuraEngine/pull/1717. It should work for 5.2

closed 2022-09-02 by nallath

PhilBaz, 2022-09-12:
@nallath 
Is there a current beta where this is fixed?  Its ruining nearly every print.