Ultimaker/Cura #12808: Support concentric not working

thmbremser, 2022-07-20:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

CL-260

### Reproduction steps

- Load a part
- set support mode to "concentric"

### Actual results

Only the corner of the model gets support. The rest of the model is not supported. V 4.12.1 does it correctly

### Expected results

Dupport in the density of the density-setting

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-07-21:
Thanks for the report.  I can't get it to work either.

Ghostkeeper, 2022-07-21:
This is something that we found ourselves as well and is listed as a bug in our backlog since May. Developers, see issue CURA-9263.

The concentric pattern doesn't work if there are any Support Wall Lines. Setting Support Wall Line Count to 0, you do get a concentric pattern (though with the wrong offset).

thmbremser, 2022-07-21:
As you can see in my screenshot, I have a line count of 1. If I increase it I only get mor wall lines at the outer side but no in the free space under the part needed to support.

Ghostkeeper, 2022-07-21:
Indeed, it only generates if the line count is 0. And that is a confirmed bug.

closed 2022-11-17 by Vandresc