Ultimaker/Cura #12813: Unexpected Slicing Issue

MrSpoon5, 2022-07-20:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

1. Opened a file for a 12 hole ocarina (https://www.thingiverse.com/thing:5411395)
2. Chose the extra fine setting
3. Made extruder 1 (PLA, printcore AA) 100%
4. Made extruder 2 dedicated to supports
5. Tried slicing, "slicing failed"

### Actual results

[image]

### Expected results

It should slice and allow me to export it to be able to 3d print.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[UM3_Updated12HocFixed.zip](https://github.com/Ultimaker/Cura/files/9153938/UM3_Updated12HocFixed.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9153951/cura.log)

GregValiant, 2022-07-20:
Thanks for the report.
I was able to load the file and it sliced fine.  When trying to preview the file Cura made it to 99% and then crashed.  Something is going on there.  I did manage to get a gcode by not switching to preview.  The crash happened 3 times in a row.
[cura log.zip](https://github.com/Ultimaker/Cura/files/9155000/cura.log.zip)

jellespijker, 2022-07-22:
Slicing with CuraEngine in debug mode also showed no crash for me. But there were a lot of warnings:
```
[WARNING] Warning: Couldn't find previous extruder plan so as to set the standby temperature. Inserting temp command in earliest available layer.
[WARNING] Empty extruder plan detected! Discarding extrusion temperature command.
[WARNING] Unnecessary extruder switch detected! SliceDataStorage::getExtrudersUsed should probably be updated.
[WARNING] Unnecessary extruder switch detected! SliceDataStorage::getExtrudersUsed should probably be updated.
[WARNING] Unnecessary extruder switch detected! SliceDataStorage::getExtrudersUsed should probably be updated.
```

The preview mode "worked" for me as well, although it is really sluggish and it shows a lot of support structures in locations where there shouldn't be any support.
[image]

Reading the earlier gcode is also really sluggish.

I will discuss this issue with the team

Vandresc, 2022-07-26:
Hi @MrSpoon5 thank you for your report!
The weird support will be addressed in Cura-9397 (devs for our internal reference), but the crash I couldn't reproduce either, neither in Prepare or Preview mode. Seeing our backlog, we believe we will not have time to address this anytime soon and will therefore defer this issue.