Ultimaker/Cura #12814: Cura 5.1.0 fails to slice in vase mode

f-engel-hrs, 2022-07-20:
### Application Version

5.1.0

### Platform

Windoes 11

### Printer

Creality Ender 3

### Reproduction steps

1. Load this STL in Cura: https://media.printables.com/media/prints/34332/stls/339390_b754e881-b09d-4de1-b7bb-c3c5e5adaf7c/9-sided-twist-planter.stl
2. Enable Spiralize Outer Contour
3. Slice and look at the preview

### Actual results

There are strange travel path that mess up the whole object. Also it seems like the nozzle travels around each layer twice

### Expected results

Vase mode prints work just fine (compare screenshots from Cura 4.13.0)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CE3_9-sided-twist-planter_PETG_03.zip](https://github.com/Ultimaker/Cura/files/9154065/CE3_9-sided-twist-planter_PETG_03.zip)
[image]
[image]

Cura 4.13.0 with the same model:
[image]
[image]

smartavionics, 2022-07-20:
Someone will undoubtedly make the comment that models have to be solid (no holes) to be printed in vase mode but that is a regression introduced in 5.x. As you know, 4.13.0 and earlier releases cope just fine with this model.

GregValiant, 2022-07-20:
_"...models have to be solid (no holes) to be printed in vase mode..."_
There ya go.
Turning on "Remove All Holes" in the mesh fixes section will allow this model to be spiralized.

Here it is sliced in 4.12 with Spiralize turned on.  That back and forth shuffle between the inside surface and the outside surface appears to be a seam but if it shows up in a print (and sometimes it does) it is in the XY plane.  The Z does spiralize.
[image]

This is 5.1.0 with Remove All Holes turned off.  Cura is able to tell that there is an outside surface facing inwards, and outside surface facing outward, and prints both.  It appears to be extruding strings across the center area.  The Z does not spiralize.  This is what is shown in the OP's images.  I agree it's a mess.
[image]

5.1.0 with Remove All Holes turned on.  The Z spiralizes up.  The bottom pattern will be concentric no matter how it is set in Cura.
I'll leave this open as a bug report for further comments but it is the new "expected behavior".

[image]

Ghostkeeper, 2022-07-21:
Duplicate of https://github.com/Ultimaker/Cura/issues/11843.

closed 2022-07-21 by Ghostkeeper