Ultimaker/Cura #12816: Slicing error

MrSag90, 2022-07-21:
### Application Version

5.0.1 

### Platform

Windows 10 

### Printer

Bear Ghost 5 

### Reproduction steps

1. I added SKL file (i tried many)
2. and get bootloop on slicing 

### Actual results

nothing, error, it says report bug here 

### Expected results

Slicing for continue printing 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

Ghostkeeper, 2022-07-21:
Could you share a project file for us? That will allow us to reproduce the exact slice that you're making, so it should allow us to reproduce the crash.

You can create a project file by going to File -> Save Project... This will save your model, settings and profiles in use.

MrSag90, 2022-07-21:
It give error with any project, even from calibration addon

Ghostkeeper, 2022-07-21:
Could you still share a project file though? If it occurs in any project, then that might mean that some post-processing or setting is causing it.

MrSag90, 2022-07-21:
[Шашка.zip](https://github.com/Ultimaker/Cura/files/9160257/default.zip)

> Could you still share a project file though? If it occurs in any project, then that might mean that some post-processing or setting is causing it.

Vandresc, 2022-07-22:
Hi @MrSag90 , unfortunately the zip contains only a.stl file which doesn't help to reproduce the issue.
Could you go to File/Save as project and attach the zip here? 
Thank you!

MrSag90, 2022-07-22:
> Hi @MrSag90 , unfortunately the zip contains only a.stl file which doesn't help to reproduce the issue. Could you go to File/Save as project and attach the zip here? Thank you!

But the issue, Cura doesnt slicing any STL file on my PC....