Ultimaker/Cura #12817: Layers not following curve, Unexpected slicing issue

blayno, 2022-07-21:
### Application Version

5.0 and 5.1

### Platform

windows 10

### Printer

Ender 3

### Reproduction steps

Sliced a file in cura 5.0 and 5.1 but the end of my print is jaggered, the layer lines are not folling the curve of the outlined stl resulting it the end of my print looking like crap, i dont have this issue in cura 4.13.
[image]
1

### Actual results

Layers not following a the curved section of the outlined stl, Layers are jaggered, cura 5.0 and 5.1 are the sam
[image]
e

### Expected results

expected Layers to follow the curve of the outlined stl like in cura 4.13.
[image]
1

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[github cura bad slice.zip](https://github.com/Ultimaker/Cura/files/9158602/github.cura.bad.slice.zip)

GregValiant, 2022-07-21:
"Windows cannot open the folder "github cura bad slice.zip" because the folder is empty."

I've never seen that one before.  It appears that the Zip folder is corrupted.  Could you try again?

blayno, 2022-07-22:
try this one bud
[cura weird layers.zip](https://github.com/Ultimaker/Cura/files/9164986/cura.weird.layers.zip)

blayno, 2022-07-22:
its extracting for me, if i click on the link and download then unzip, my file is in there, not sure what the go is there

GregValiant, 2022-07-22:
That worked.
The top of the model comes to a knife edge.  Those are always tough as the sharp point going over the top gets down below a "Line Width" thick.  There is a slight radius on the backside that is also having an effect.
You can adjust the Walls setting "Split Middle Line Threshold" to 50% and the slice will get better.  There just isn't much cross section width at that point on the model and so the line width has to be adjusted down a bit more.
I don't really see a bug here but the Cura Team will take a look.

blayno, 2022-07-22:
thank you for the tips,although still not as good as cura 4.13.1, i corrected it mostly using print thin walls and split middle line threshold @ 1%
[image]

blayno, 2022-07-22:
i further increased the quality by using "Add middle line threshold " @ 10%, Looking good now
[image]

Vandresc, 2022-07-26:
Hi @blayno, it seems we already have a related ticket in our backlog. Will share this issue in the ticket as well. Thank you for your feedback. 
Devs, see CURA-9399 for out internal reference.

jwhite, 2023-04-04:
This is destroying me.  "Add middle line threshold " seems to be missing in version 5.3.0 -- so there is no workaround.

5axes, 2023-12-06:
**Minimum Wall Line Width** 
[image]