Ultimaker/Cura #12818: .3mf File problem

Terrybi, 2022-07-21:
### Application Version

5.0.0 and 5.1 Beta and 5.1

### Platform

Win 11

### Printer

Creality Ender 3 pro

### Reproduction steps

1. loaded .3mf file into cura
2. file loaded in left front corner
3. moved model to center of build plate in cura
4. saved gcode file
5. attempted print
6. model printed in front left corner.
7. stopped print job
8. opened model in 3dbuilder and saved as an stl
9. loaded stl in Cura 5.0.0
10. model loaded in center of build plate
11. saved gcode and printed in center of build plate with no problem at all

### Actual results

1. after a few print attempts I uninstalled all cura versions and reinstalled 5.0.0 and removed all Cura files from computer
2. reinstalled Cura 5.0.0 clean install
3. retried steps above with no improvement
4. retried steps above only with same model as an stl file and model print worked as expected.

### Expected results

* NOTE* I have noticed there are other users that appear to have reported similar problems with 3mf files but I wasn't sure if I should just add mine to theirs or just it this way
[heelcover.zip](https://github.com/Ultimaker/Cura/files/9159098/heelcover.zip)
.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

 I had a problem with a 3mf file a while ago using 5.0.0 where I was unable to use support blocker successfully but I no longer have that file.

MariMakes, 2022-08-24:
Hey @Terrybi,

Thanks for opening your first issue with us! 🎉 

I've taken a look but I have trouble understanding your issue. 
You see that the .3mf that you load appears in the front left of your build plate but you expected it to appear in the center? 

One of the added benefits of a .3mf is that it remembers its location.
You can export your file to an .stl and it will forget its location on the build plate or you can re-save your file as a .3mf after you placed it in the correct location.
https://user-images.githubusercontent.com/40423138/186466988-edf787f1-8dcc-461b-9f70-9b357b78ecf7.mp4

If that is not your issue. Can you please formulate your issue as:
**I see ... but I expected ...**

That helps me help you! 👼

Terrybi, 2022-08-24:
Good day! Thank you very much for your reply with the mp4 demo. I wish I 
had of known this before, that was so easy once you explained it. I've 
seen some people with the same, or similar problem and nobody was able 
to point any of us in the right direction like you just did!

  I tried a couple of my 3mf files and followed you direction and all 
worked perfectly! Thank you once again, I'll be using 3mf files a lot 
more now.

Terry.

MariMakes, 2022-08-25:
Happy to help. 😺 
I'm closing this issue since it's a user error.

closed 2022-08-25 by MariMakes