Ultimaker/Cura #12819: Infill does not update relative to menu selection

jmccurley-lumco, 2022-07-21:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker S5 Pro

### Reproduction steps

1. Import Part and Set Up for Printing (speeds, materials, support, etc.)
2. Change infill type (I usually go with Gyroid, but upon encountering issue, I have tried almost every option)
3. Slice Part
5. Print Part

### Actual results

After Slicing, preview shows Triangle infill, regardless of actual infill selection.  I have had this in Cura before, but it printed correctly, so I printed anyway, thinking it was a visual bug.
Part Prints with triangle infill, even though Gyroid was selected before slicing.

### Expected results

Part should preview and print with Gyroid if Gyroid is selected in the settings.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[UMS5_M11482-D116.zip](https://github.com/Ultimaker/Cura/files/9159490/UMS5_M11482-D116.zip)

GregValiant, 2022-07-21:
Thanks for the report.
I have heard of this bug before but I have never been able to reproduce it (Win 10 and currently Cura 5.1.0)
This is your project just switched to gyroid with no other changes.
Someone from the Cura Team will take a look.

[image]

GregValiant, 2026-08-19:
The current Cura version (5.13.0) doesn't seem to exhibit this behavior.  I'll close this.

closed 2026-08-19 by GregValiant