Ultimaker/Cura #12821: Matt - Slicing error

OntelPD3D, 2022-07-21:
### Application Version

5.0.0

### Platform

MacOS Monterey

### Printer

Ultimaker S5

### Reproduction steps

There was a slicing error while preparing to print a part file.

### Actual results

I have to try slicing again.

### Expected results

I'm hoping I can print successfully.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_AA Portable Fan_Front.3mf.zip](https://github.com/Ultimaker/Cura/files/9161114/UMS5_AA.Portable.Fan_Front.3mf.zip)

closed 2022-07-21 by OntelPD3D

GregValiant, 2022-07-21:
Thanks for the report.
5.0.0 had trouble with some models and I think the features of your model are triggering that bug in Cura 5.0.
When I open the project file in 5.1.0 the slicing error popped up.  When I only imported the model into 5.1.0 then it would slice.
My take on this is that something carried over in the 5.0 project file that keeps that model from slicing, but the model itself is fine.  I would suggest upgrading to Cura 5.1.  There is a workaround for 5.0 and that is to rotate the model 45° on the build plate.  It should slice like that.
Someone from the Cura Team will look at this.

OntelPD3D, 2022-07-22:
Thanks for the feedback. Please also note, when I adjusted some of the print parameters, specifically the print stepping (I think to 1.5mm instead of 2.0mm), it successfully sliced. I'll upgrade Cura to 5.1 and hope this takes care of it all together. Thanks.