Ultimaker/Cura #12824: Rotating scene stuck, interface not responding.

themanro, 2022-07-21:
### Application Version

5.1.0

### Platform

macOS Monterey

### Printer

Qidi X-Max

### Reproduction steps

Get object into Cura, work on settings, trying to rotate the scene holding "Control" key and using tablet (wacom), scene rotates and won't stop after key released.

### Actual results

Interface becomes unresponsive, just keeps rotating around as I it would if I had a hot key pressed. Lot of times interface won't respond to tablet clicks at all. I have to constantly keep restarting the app.

### Expected results

It only rotates when the hot key is pressed. All interface buttons respond to tablet at all times.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9162992/cura.log)

Vandresc, 2022-07-25:
Hi @themanro, did you also have these issues with 5.0 or 4.13?

themanro, 2022-07-25:
> Hi @themanro, did you also have these issues with 5.0 or 4.13?

It has started happening with 5+ release.

MariMakes, 2022-08-25:
Hey @themanro 
Thanks for your report 🎉 

This seems to be a duplicate.
I'll close this issue as a duplicate, you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12554

closed 2022-08-25 by MariMakes