Ultimaker/Cura #12825: Version Main instead of version 5.1

ckvsoft, 2022-07-21:
### Application Version

5.1

### Platform

Linux Appimage

### Printer

any

### Reproduction steps

Slice any file. 
Then in the gcode
;Generated with Cura_SteamEngine main
instead of the version number as before

### Actual results

;Generated with Cura_SteamEngine main

### Expected results

;Generated with Cura_SteamEngine 5.1.0

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

;FLAVOR:Marlin
;TIME:22875
;Filament used: 55.373m
;Layer height: 0.16
;MINX:112.325
;MINY:121.379
;MINZ:0.2
;MAXX:225.675
;MAXY:208.619
;MAXZ:231.24
;Generated with Cura_SteamEngine main
M82 ;absolute extrusion mode

jellespijker, 2022-07-22:
Hi @ckvsoft this is indeed a bug, thank you for reporting this, we will pick this somewhere in the next two weeks, so it should be fixed for our 5.2.0 release.

Devs see CURA-8830

ckvsoft, 2022-07-22:
thanks. It's not a huge bug. Just wanted to mention it