Ultimaker/Cura #12826: Overlap settings broken

guate6, 2022-07-22:
### Application Version

5.1.0

### Platform

Windows 10/11

### Printer

Ultimaker S5

### Reproduction steps

Placed custom block on solid model to modify settings for overlap. Tried both "infill mesh only" and "cutting mesh" 
Main model is 20% infill, but a specific space I wanted it to be 40% - both cubic subdivision. 

### Actual results

Weird interface at the base of the custom cube, and rest of model where it should be 40% is 0% infill

### Expected results

Should work as it did in version 4.x

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Although these images/video are of one or two models, I've repeated this with several solid models and the bug is the same. 

[image]
[image]

[image]
[image]
[image]

[image]
[image]

video capture: https://www.dropbox.com/s/39dkteg8cn5zudj/Body%20%2B%2010in%20slots%20-%20Ultimaker%20Cura%202022-07-21%2020-51-09.mp4?dl=0

GregValiant, 2022-07-22:
Thanks for the report.
I could not get it to work with Cubic Subdivision either.  It looks like the Mesh Modifier goes to 0% infill density?  A workaround is to pick a different type of infill.
The Cura Team will look at this.

guate6, 2022-07-22:
I need strength in all 3 axes. Do you have a recommendation?

GregValiant, 2022-07-22:
Most infills just hold up the ceiling and provide about zero structural support.  Gyroid might be the best example and people love it, but it has no strength and is more like an accordion bellows than anything else.
When I want strength I go to "Grid" at 15-20% density, set the "Infill Line Multiplier" to 3, and set the "Line Directions" to be at 90° to the walls (the default is [45, 135] and this is [90]).
This is at 15% density.  At triple width - each of the interior grid lines is a rib 1.2mm thick spaced about 4mm apart.   You could park a car on this.  I have found that using triple width instead of double width prints a lot smoother.
[image]

guate6, 2022-07-22:
I'll give your line multiplier suggestion a try.
Grid, according to the ultimaker site, is not 3D (3-axis) support, so I'll
give one of these others a go until the bug gets fixed.

Thank you.

https://support.ultimaker.com/hc/en-us/articles/360012607079-Infill-settings

GregValiant, 2022-07-25:
When Grid is at the default line directions of [45, 135] then it is only strong when the load is top-to-bottom.  It's just an accordion if the load is front-to-back or side-to-side.  With the line directions at [0, 90] (as in my previous image) then it's just as strong from any direction when any load is applied at 90° to the surface.

guate6, 2022-07-25:
So to ensure three-dimensional strength you're suggesting I change it to
[0, 90]?

GregValiant, 2022-07-25:
For a rectangular shape that is square to the bed then yes, [0 , 90] works.  If you need to rotate a print 45° so it fits the build plate then maybe you want the line directions at [45, 135].  There is no hard and fast rule that says "Always do it like this".
If the shape of the exterior of the print was a hexagon then that would require something different as well.
When the ribs are square to the exterior walls it's about as good as it will get.  The more robust the ribbing then the stronger the print.  An internal construct that is .4mm wide isn't going to give much (if any) strength.  At 1.2mm wide, that construct becomes a rib, gusset, whatever you want to call it.  It becomes much stronger in compression when the loading is the same vector as the length of the rib.  That's the best you can do.
You can add support blockers and configure their "Per Model Setting" as as "Modify Settings for Overlaps" and as an Infill Mesh.  Then you can change the line directions and infill density for the areas where the blocker intersects the model.  And blockers can be sized and moved like any other mesh.  There are alternate shapes available in the Cura MarketPlace.

Vandresc, 2022-07-26:
Devs see CURA-9517 for our own internal reference. 
Thank you for your report and @GregValiant thank you for your workaround.