Ultimaker/Cura #12827: Longer Layer Seek Bar

seanbrockest, 2022-07-22:
### Is your feature request related to a problem?

They layer seek bar is barely 1/3 the height of the window. In prints with several hundred layers, there isn't enough fine control to seek for individual layers, except for typing them in, which takes a lot of time.

### Describe the solution you'd like

They layer seek bar needs to be much longer, there's no reason it cannot be 95% the height of the window, allowing for much more fine control when seeking layers. Optionally you could allow us to custom scale the seek bar ourselves, so people who don't want a long seek bar won't complain.

### Describe alternatives you've considered

Described in earlier sections.

### Affected users and/or printers

All users of Cura

GregValiant, 2022-07-22:
Thanks for the thought.
You may know this...You can type a layer number into the text box (that displays the layer number of the slider) and then hit Enter.  With the cursor in that textbox you can use the Up/Down arrow keys on your keyboard to scroll up and down through the layers.

Vandresc, 2022-07-22:
Hi @seanbrockest, exactly what @GregValiant said, I'd suggest using the keyboard to move up and down through layers for finer control. I brought this up with our UX designer and seeing what a busy backlog we have, we are not going to implement this any time soon; therefore, we will defer. We do want to thank you though for thinking of enhancements with us.

closed 2023-08-06 by github-actions[bot]