Ultimaker/Cura #12828: 5.1.0: Strange segmentation

Sineos, 2022-07-22:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

1. Slice model https://www.printables.com/de/model/153547-awesome-bee-feeder --> The hex grid
2. Look at the resulting gcode in a gcode viewer

### Actual results

Segmentation in certain areas seems unnecessarily granular:

[image]

### Expected results

Uniform segmentation

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE5_Grid.zip](https://github.com/Ultimaker/Cura/files/9165989/CE5_Grid.zip)

GregValiant, 2022-07-22:
Thanks for the report.
It's a geometry thing and it has to do with your line width.
A group of three adjoining hexagons leaves a triangle shaped area.  The dimension from the mid-point of that triangle shaped area to any of the adjoining points of a hexagon is greater than the distance between the hexagons.  So Cura is stopping to fill that void with a quick circle movement.  In your model the land between the hex openings is 2mm but the triangle gap between the hexes is 2.3mm.  It's that 0.3mm hole that is causing the tool path to be what you see.
If you change your line width down to .40 you can see how the filling of that area changes, but at .4 line width the print time goes way up.  Setting it to .5 line width gives better results time-wise and the tool path is smoother.
I don't see a bug here but rather a different way of doing things.  We used to be able to turn off "Fill Gaps Between Walls" and avoid this.  Now we play with the line width.
One of the folks on the Cura Team will take a look.

Sineos, 2022-07-22:
Many thanks for the explanation. Really appreciated. Time to get a handle on the new slicing strategy.
Only thing throwing me off, is the fact that I can perform well, e.g.

[image]

Also considering the workflow it sounds a bit strange to have to load a sliced gcode file into https://gcode.ws/ to see if anything strange happened that may throw off the printing process later on.
E.g. my personal experience so far is that I get a visually more pleasant part with .42 line-width compared to .48 or .5

GregValiant, 2022-07-22:
Part of this is the visualizer software.  Cura's is written one way, that one you use is written another way.  The one below is written a third way.  Each is a different solution to the same problem by different software engineers.  The one thing I'm sure about is that my printer will do it's own thing.  It prints very well so I'm good with that.  My laptop is in Compatibility Mode so I don't get the fancy graphics BUT the flat line view shows the little circle that Cura is making when it puts down the dots at the intersections.
(BTW...I think "Visually Pleasant" would be more relevant from a  bees eye view?)
[image]

Sineos, 2022-07-22:
> (BTW...I think "Visually Pleasant" would be more relevant from a bees eye view?)

Might be unique to my printers but I found that an extrusion width of .42 produces smoother wall stacking and top surfaces

I played a little with the extrusion width:
* .43: Makes it even worse. The lines with only 2 segments as shown in screenshot above now have multiple segments
* .44: About the same as .42
* .45: About the same as .43
* .46: About the same as .45
* .47: Starting to get better, some additional segments on the once (.42) perfect lines but way less on the densely segmented ones
* .48: According to my expectation
* .49 / .50: Like .48
* .51: Even less segments

I may not have fully understood the correlation between extrusion width and number of segments on a straight line but it seems strange that proper results can only be obtained at values .48 and above.

Vandresc, 2022-07-25:
Hi @Sineos, I will bring this up with the rest of the team, there is indeed something weird happening there. 
Until then I think we have a workaround for you, set the Maximum Extrusion Area Deviation to 200000, in your profile is set to 2000. That will remove the extra segments we are seeing.

Sineos, 2022-07-25:
Thanks @Vandresc. Your efforts and Cura in general are highly appreciated.
I can confirm that even setting the `Maximum Extrusion Area Deviation` to just 10000 dramatically improves the situation:

[image]

What is considered a "sane" default for this setting? As I never touched it, 2000 appeared to be default after installing 5.1

Vandresc, 2022-07-29:
The team agrees that we will need to investigate this, a ticket has been created on our backlog. Devs see CURA-9524 for our internal reference.
@Sineos I think 50.000 is a "safe" value.

commit 03c88c18 (2022-08-24): Bump up the maximum_extrusion_area_deviation

CURA-9524
Fixes #12828
closed 2022-08-24 by nallath