Ultimaker/Cura #12831: Missing release description

m-sundman, 2022-07-22:
### Is your feature request related to a problem?

A new release variant has been introduced (`5.1.0-linux-modern`), but there is no description of what this is and how it differs from the `5.1.0-linux` variant.

### Describe the solution you'd like

Descriptions for all variants released.

### Describe alternatives you've considered

It might be possible to download both variants, and run them and try to figure out what the difference is.

### Affected users and/or printers

All linux users are affected.

ckvsoft, 2022-07-22:
modern is for newer/current Linux distributions

jellespijker, 2022-07-22:
Although the changelog mentioned this as @ckvsoft mentioned; I think this is a fair point. I will discuss with the team how we can better convey this information. Or better yet, how we can go back to 1 single AppImage again.

m-sundman, 2022-07-22:
> modern is for newer/current Linux distributions

What is "newer/current"? What isn't?

Ghostkeeper, 2022-07-22:
The "modern" version requires glibc 2.31 or higher. Functionally, the difference is that it's built using the system libraries of Ubuntu 22.04 instead of 20.04.

This is not commonly a problem with any other software as far as I know. We hope that this will be a temporary solution. But for communication it's very hard to write "Cura for Glibc 2.31+" because hardly anyone knows what that even is or how to find the version number.

I hope that we find a solution to the compatibility problem soon and/or get a help article about it online.

m-sundman, 2022-07-24:
> The "modern" version requires glibc 2.31 or higher.

Ah, this description inside the release notes would be enough for me, thank you!
(And for anyone reading this wondering what libc they have, if you're using a debian-based linux, which is what's most common, then you can probably run this in the terminal: `dpkg -s libc6` and check the version line.)

m-sundman, 2025-05-09:
Seems fixed.

closed 2025-05-09 by m-sundman