Ultimaker/Cura #12832: Circular prints 2nd brim line steps back?

andersp5, 2022-07-22:
### Application Version

5.1.0 & 5.0.0

### Platform

Windows 10

### Printer

Ender 3v2

### Reproduction steps

On a circular print with a brim, after the very first brim line the printer steps back then prints the second line. On v4.13 the same print the 2nd line step across from the first and doesn’t step back

### Actual results

As the printer steps back filament is still being ejected and this causes the head to block slightly and thus this line fails to stick to the bed properly. You can see this effect by using the visual video monitor in Cura very clearly you don’t have to actually print the part to see this issue. Look at v4.13 and then v5.1 there is a difference with this stepping back issue. Circular prints seem to be the worst offenders. I have tried to find a difference in the profiles but with no success yet but it might just be a setting causing this that I am unaware of in V5 onwards perhaps?

### Expected results

As in v4.13 I expected the 2nd line to step from the 1st line and not step back before continuing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Will upload file if required at a later date

Vandresc, 2022-07-25:
Hi @andersp5 , this is indeed a known issue to us and we have a ticket on the backlog. Devs see CURA-9087 for our internal reference. Thank you for your feedback.

andersp5, 2022-07-25:
Hi Vandresc

Sorry I must have missed anything relating to this issue when I checked but glad to hear you have it in hand and thanks for the heads up. 

At the moment I have unfortunately reverted back to V4.13 but look forward in the future to fully move to version 5 when this is fixed. By the way it is still present in version 5.1.0

The Cura slicer is great by the way and I very much apricate all your efforts and developments in providing us with the latest V5, with the new engine, which seems to work very well from my limited experience so far.

Thank you again

From: Vandresc
Sent: 25 July 2022 09:52
To: Ultimaker/Cura
Cc: andersp5; Mention
Subject: Re: [Ultimaker/Cura] Circular prints 2nd brim line steps back? (Issue#12832)

Hi @andersp5 , this is indeed a known issue to us and we have a ticket on the backlog. Devs see CURA-9087 for our internal reference. Thank you for your feedback.
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

andersp5, 2022-08-02:
Hi Vandresc,

Regarding this issue I have been checking a few files and discovered that it is not all of them?  Strange this one!!

Therefore I thought the attached files may help you locate what the difference may be. 

The “Motorhome Shower lighting on-off button holder ver2.stl” with a brim jumps back after the first line of the brim for the start of the second line but the “Windmill Back Carrier Washer V1.stl” with exactly the same brim and profile does not? 

The only difference that I can see is the size of the circle diameter and the height of each item, there is nothing else as the profile used is exactly the same, which I have also attached.

Of course it could be a coding problem deep in the programme but this may assist you to narrow it down perhaps.

Regards

andersp5

From: Vandresc
Sent: 25 July 2022 09:52
To: Ultimaker/Cura
Cc: andersp5; Mention
Subject: Re: [Ultimaker/Cura] Circular prints 2nd brim line steps back? (Issue#12832)

Hi @andersp5 , this is indeed a known issue to us and we have a ticket on the backlog. Devs see CURA-9087 for our internal reference. Thank you for your feedback.
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Vandresc, 2022-08-02:
Thank you @andersp5, will make sure to attach this scenario to our ticket as well.

Vandresc, 2022-08-02:
Looks also like a duplicate of https://github.com/Ultimaker/Cura/issues/11936

andersp5, 2022-08-02:
Yes it does, obviously it is causing more problems than I first thought

From: Vandresc
Sent: 02 August 2022 13:09
To: Ultimaker/Cura
Cc: andersp5; Mention
Subject: Re: [Ultimaker/Cura] Circular prints 2nd brim line steps back? (Issue#12832)

Looks also like a duplicate of #11936
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus