Ultimaker/Cura #12833: Auto generated comment in G-Code has no version info on Cura 5.1.0 for Windows

Tiryoh, 2022-07-22:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Kingroon KP3S

### Reproduction steps

1. Install Cura 5.1.0 on Windows
2. Slice a STL (any STL and any config is OK)
3. Save G-Code
4. Open G-Code with text editor and check the auto generated comment

### Actual results

```
;Generated with Cura_SteamEngine main
```

[image]

### Expected results

```
;Generated with Cura_SteamEngine 5.1.0
```

For the reference, Cura 5.0.0 shows the version correctly.

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Once this issue is fixed, an application like Klipper can get the G-Code info correctly.

[image]

GregValiant, 2022-07-22:
Thanks for the reporrt.
This is a duplicate of #12825 and I'll mark it as such and close this in favor of the previous report.  If you want to leave more comments you can put them in that other report.
(BTW...Nice eye.  I looked at _a lot_ of gcode files sliced with 5.1 and missed this completely.)

closed 2022-07-22 by GregValiant

Tiryoh, 2022-07-22:
Oops, sorry for creating the duplicate issue and thanks for checking this issue.
I'll continue checking the update on https://github.com/Ultimaker/Cura/issues/12825.