Ultimaker/Cura #12834: Font Color in Dark Mode

MarioFrom98, 2022-07-22:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Sidewinder X2

### Reproduction steps

I switched to Dark Mode

### Actual results

The font remained light gray mainly for Projectnaming and object list
[image]

### Expected results

The font color for dark mode should be a complementary color to the grayish background. Maybe blue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It isnt extremely urgen

Toddi3D, 2022-07-23:
it would be good if you could choose from four or five colors. I would like to take orange

Vandresc, 2022-07-26:
Hi @Toddi3D , thank you for your report!
Devs, see CURA-9315 for our internal reference.

nallath, 2024-04-05:
This has already been fixed

closed 2024-04-05 by nallath