Ultimaker/Cura #12835: Cura generate tree supports in the air

Mascantosh, 2022-07-23:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender 5 

### Reproduction steps

All information in attachment project

### Actual results

[image]

### Expected results

Airflow parts connected with bottom plate

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Airflow_bug.zip](https://github.com/Ultimaker/Cura/files/9172727/Airflow_bug.zip)
In attached project
Go to -> **Preview** and select 165 layer

DerGenaue, 2022-07-23:
Duplicate of #12326

Vandresc, 2022-07-25:
Indeed a duplicate, we will close this ticket. @Mascantosh pls check the duplicate for progress. Thank you!

closed 2022-07-25 by Vandresc