Ultimaker/Cura #12836: (5.1.0) Cannot change materials or start slicing

bellzw, 2022-07-23:
### Application Version

5.1.0

### Platform

65-bit Windows 10 on a Lenovo Yoga

### Printer

BIBO (with my own printer definition)

### Reproduction steps

1. Launch Cura
2. Put calibration cube on surface
3. Put anti-warp tabs at corners
4. Try to slice
5. Try to change materials
[image]
[image]

### Actual results

1. Cura launches
2. Cube appears
3. Tabs appear at corners
4. Bar appears under materials, but clicking on it results in no response.  No list of materials appears.
5. Black bar appears over Slice button, but no slicing occurs.  Blue button changes color, but is not responsive.

### Expected results

1. Cura launches (this happens)
2. Cube appears (this happens)
3. Tabs appear at corners (this happens)
4. List of materials from which to select should have appeared.  All the bundled materials are present, and I've imported a few of my own in the list
5. Slicing should have started

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This is the printer definition file I am using.  The file included with Cura uses fdmprinter as the default and this file has speeds and accelerations that do not apply to the BIBO.  Time estimates are withing a few % with mine, off by a factor of 2 or more with the default.

[bibo2_dual_zwb_startup_feedrate.def.txt](https://github.com/Ultimaker/Cura/files/9172957/bibo2_dual_zwb_startup_feedrate.def.txt)

GregValiant, 2022-07-23:
Thanks for the report.
Before I get into this, you have added the necessary Extruder files?

bellzw, 2022-07-23:
Yes, I had added the extruder files to the correct folder, although I had to search for that folder since the convention is not the same as 4.12.1, from which I updated.  I’m sure it’s in some documentation I haven’t yet read.

 

The problem appears to have resolved itself after exiting and restarting 5.1.0.  I thought it might have been related to adding the settings guide plug-in to my installation, something I hadn’t investigated late last night (EDT), so I removed it and restarted.  It’s now responding correctly to let me change materials and to slice.  I reinstalled the settings guide and added the Extensive Support Logging plug-ins to my installation in case something like this or some other odd behavior happens again.  It’s still working after adding those back, so I guess you can mark this as a false alarm, closed, or resolved, as appropriate.

 

Sorry if I’ve caused unnecessary work.

 

Zane

GregValiant, 2022-07-23:
No trouble at all.
As long as you got this sorted out I'll go ahead and remove the bug label and close it.

closed 2022-07-23 by GregValiant