Ultimaker/Cura #12837: I request you stop ruining sidebar gui!

Katoom990, 2022-07-23:
### Is your feature request related to a problem?

I much prefer the old layout like sidebar gui offers but so many of your upgrades remove that feature.  Please make it a permanent option again. 
I really can't stand to have my buildplate image covered by the settings menu. I always leave my settings open as I tend to make a lot of changes; experimenting with different materials and difficult prints.

### Describe the solution you'd like

Make Sidebar Gui a permanent option instead of a plugin please.

### Describe alternatives you've considered

The only other solution is to hope whomever makes that plugin keeps motivated with updates.

### Affected users and/or printers

Many people prefer this or it wouldn't be a plugin.

fieldOfView, 2022-07-23:
Hi, author of the SidebarGUI here. Thanks very much for the vote of support for my work.

Cura does not "ruin" the Sidebar GUI, but a choice was made to take the GUI in another direction than the sidebar GUI. What works well for you may not be the best solution for everybody else (and vice versa). Though I personally agree with you that the UI provided by the Sidebar GUI  Plugin is a better experience, the UX designers have put a lot of thought in what is the current design. 

Maintaining two different UIs is an unnecessary amount of work, so having what the sidebar gui plugin gives you as a standard option is ~unlikely~ probably not going to happen.

Also see https://github.com/fieldOfView/Cura-SidebarGUIPlugin/issues/31, which by the way has a prerelease of the next update of the plugin which is compatible with Cura 4.0 - 5.1

jellespijker, 2022-07-29:
@Katoom990 I just approved the new release for the sidebar GUI plugin. Once @fieldOfView publish it you will once again be able to enjoy the sweetness of this plugin.

I see that our UXer @lorenzoromagnoli is tagged, so I will leave this issue open for him to decided.

Katoom990, 2022-07-31:
Thank you!  
So much better imo

Chris

lorenzoromagnoli, 2022-08-01:
Thanks for your request @Katoom990, I agree with you and Aldo that the old sidebar or the sidebar plugin provides a better experience for advanced users. Unfortunately, as much as I'd like to bring the sidebar back to Cura there are so many things we would like to get to that I'm afraid the sidebar won't come back very soon. 

Enjoy @fieldOfView 's plugin for now and never stop thanking him for his effort in maintaining his plugins and making Cura a better place :)

closed 2022-08-01 by lorenzoromagnoli