Ultimaker/Cura #12838: DeprecationWarning

betterwithage, 2022-07-23:
### Application Version

Ultimaker-Cura-5.1.0-linux.AppImage

### Platform

Linux 64bit

### Printer

renkforce rf100xl

### Reproduction steps

starting cura on the console gives the following message:

PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Cyclic dependency detected between "file:///tmp/.mount_cura0vYVyA/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_cura0vYVyA/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext

### Actual results

cura starts, no more errors or warnings.

### Expected results

no warnings

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9173385/cura.log)

jellespijker, 2022-07-25:
Thank you for bringing this to our attention. It would be ideal if Cura would run without any warnings, unfortunately that will probably never be the case. The warnings are there to inform the user/developer of something "out-of-the-ordinary" happening, which might have a negative side affect and could require either the user or us to take action.

The deprecation  warning you're referring to will have to be taken care of once we start using Python 3.12. Since Python 3.11 is still in development phase this won't be a problem for the foreseeable future. The most pragmatic thing to do for us to handle the deprecation of the imp module when we're actually work on upgrading to Python 3.12.

Silencing the warning, might also be unwanted since we should be aware of it.

I hope you understand that I will close this ticket for the above mentioned reasons.

If you don't want to see the warnings on your shell you can redirect them to `/dev/null`
```bash
./Ultimaker-Cura-5.1.0.AppImage &>/dev/null
```

betterwithage, 2022-07-25:
Thank You for your prompt comment, I had already thought so.
Best regards, 
betterwithage