Ultimaker/Cura #12839: Bad generated support structure

65643243, 2022-07-23:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. object with several horizontal steps which need support and overhangs greater than 45 degrees and less than 90 degrees
2. new standard 5.1.0 PVA support settings

### Actual results

1. several small support strands are produced next to the actual structure. these can get caught in the print and worsen the appearance. 
2. overhangs greater than 45 degrees and less than 90 degrees are supported with poorly formed support roofs
3. concentric support roofs are generally produced worse, often with patterns that have gaps or are irregularly shaped in some other way. other support roof patterns sometimes also have this problem.

### Expected results

support leaves an unusually large number of bad spots, which makes this version poorly usable. 
reliability of the new support structure has also deteriorated. all objects with more complex overhangs than 90 degrees are recorded as misprints

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]
[2xArm_Settings2.zip](https://github.com/Ultimaker/Cura/files/9173672/2xArm_Settings2.zip)

65643243, 2022-10-06:
additional information: it seems that the problem is mainly with the topmost support interface layer. there are several places where the outermost layer is not supported. 

This can currently also be observed in the 5.2 beta. 

Both in 90° overhangs and in shallower overhangs. 
[image]
[image]
[image]

GregValiant, 2022-10-06:
I looked at this on the OP's post on the UM Forum and I can duplicate the issue with his printer as well as my own.
Changing the Initial Layer Height from .27 to .28 fixed the problem and the gaps in the top interface layer went away.